lorabasics/basicstation #110: Cannot connect to device

linhuaian, 2021-06-19:
Hello! 

I have been trying to connect my gateway and device to AWS IoT Core via basic station.
I have done until the final steps of connecting my Sensor to AWS via my gateway.

Here is some info:
Gateway: RAK7243 running on Raspberry Pi
Sensor: RAK7205

Everything seems to be fine until the point where I couldn't send uplink data to 
AWS for my sensor.

This is the error log:

2021-06-18 09:54:33.949 [AIO:XDEB] [3|WS] > {"msgtype":"jreq","MHdr":0,"JoinEUI":"60-C5-A8-FF-F9-23-72-05","DevEUI":"60-C5-A8-FF-FE-75-FE-B5","DevNonce":751,"MIC":1999970882,"RefTime":0.000000,"DR":2,"Freq":923200000,"upinfo":{"rctx":0,"xtime":69524345096022996,"gpstime":0,"fts":-1,"rssi":-61,"snr":13.5,"rxtime":1624010073.949095}}
2021-06-18 09:54:34.471 [AIO:XDEB] [3|WS] < {"msgtype":"dnmsg","DevEui":"00-00-00-00-00-00-00-00","regionid":7,"dnmode":"updn","dC":0,"diid":46829,"pdu":"209a24d7845efee0b7e6c92ccf354c00f2f15d0516e587dda261a889f0ddae511f","priority":1,"RxDelay":5,"RX1DR":2,"RX1Freq":923200000,"xtime":69524345096022996,"rctx":0}
2021-06-18 09:54:34.471 [S2E:WARN] Unknown field in dnmsg - ignored: regionid
2021-06-18 09:54:34.471 [S2E:WARN] ::0 diid=46829 [ant#0] - Tx job too far ahead: 6h50m
2021-06-18 09:54:35.585 [any:XDEB] RX mod=LORA f=922800000 bw=125 sz=97 dr=2 72FEFB28AFD29E1B62E26AB8687964E221E0B89CD3099B80173394C21D025799FD8D357FCE041C586F40943F07F87A01404DE5B60F84DFA7D97E1AD5FC515758966AAA2772B6A0C3A24AEFCFD3C357700AC9423CB0D41A4001A2920B782C137A15
2021-06-18 09:54:35.585 [any:XDEB] Dropped frame without CRC or with broken CRC

As seen above, the join request from my sensor was detected from the gateway, however it seems that the join request was not accepted by my gateway with the error message: Tx job too far ahead.
(Was wondering if this issue could be caused by timezone issue? The region set in AWS was in USA but I am currently in Singapore.)

May I know how I could have fixed this issue?

Thank you in advance!

linhuaian, 2021-06-28:
The problem was solved by adjusting the system time in Raspberry to align with the server time.

closed 2021-06-28 by linhuaian

Devansh-Shah-5400, 2022-09-27:
What about the Unknown field in dnmsg - ignored: regionid? I am currently experiencing the same problem and my device is sending requests but gateway isnt accepting them.

BTW the Unknown field in dnmsg - ignored: regionid is the only error I get.