lorabasics/basicstation #112: CUPS Protocol error. Closing connection.

ARPIIoT, 2021-06-25:
Hardware : Rpi 3B + Lora 868Mhz connected on GPIO with SPI interface

I am trying to connect my Gateway with AWS but having issue in CUPS Protocol.

[CUP:VERB] Retrieving update-info from CUPS https://XXXXXX.cups.lorawan.eu-west-1.amazonaws.com:443...
[CUP:ERRO] **CUPS Protocol error. Closing connection.**
[CUP:INFO] Interaction with CUPS failed - retrying in 1s
[TCE:INFO] Starting TC engine
[TCE:ERRO] No TC URI configured
[TCE:INFO] Router rejected or retry limit reached. Invoking CUPS.
[TCE:INFO] Terminating TC engine
[CUP:INFO] Starting a CUPS session in 1 seconds.
[CUP:INFO] Starting a CUPS session now.

**Using default station.config from _examples/live-s2.sm.tc_**

I am following below tutorial.
https://www.youtube.com/watch?v=6-ZrdRjqdTk

richardorydowner, 2021-08-13:
I have the same issue

plmwd, 2021-12-14:
I also have the same issue.

plmwd, 2021-12-15:
The issue for me was an EUI mismatch. Basic station is not calculating it how the RAKWireless calculates it. When I tried using the station id (EUI) printed from starting the basic station, AWS gave me an error saying the EUI was already taken. This is the command RAK Wireless uses to calculate the EUI:

```sh
do_get_gw_id()                                                                                                                                                                                                
  {                                                                                                                                                                                                             
      GATEWAY_EUI_NIC="eth0"                                                                                                                                                                                    
      if [[ `grep "$GATEWAY_EUI_NIC" /proc/net/dev` == "" ]]; then                                                                                                                                              
          GATEWAY_EUI_NIC="wlan0"                                                                                                                                                                               
      fi                                                                                                                                                                                                        
          if [[ `grep "$GATEWAY_EUI_NIC" /proc/net/dev` == "" ]]; then                                                                                                                                          
          GATEWAY_EUI_NIC="usb0"                                                                                                                                                                                
      fi                                                                                                                                                                                                        
                                                                                                                                                                                                                
      if [[ `grep "$GATEWAY_EUI_NIC" /proc/net/dev` == "" ]]; then                                                                                                                                              
         echo ""                                                                                                                                                                                                
      fi                                                                                                                                                                                                        
      GATEWAY_EUI=$(ip link show $GATEWAY_EUI_NIC | awk '/ether/ {print $2}' | awk -F\: '{print $1$2$3"FFFE"$4$5$6}')                                                                                           
      GATEWAY_EUI=${GATEWAY_EUI^^}                                                                                                                                                                              
      echo $GATEWAY_EUI                                                                                                                                                                                         
  }
```

[You can then set the EUI](https://doc.sm.tc/station/conf.html#station-identity) in `station.conf` : 
```json
...
"station_conf": {
    ...
    "routerid": "<EUI>",
}
```

If that doesn't work for you, I've attached a patch that prints the response body when basicstation errors out due to a CUPS protocol error
[debug-patch.zip](https://github.com/lorabasics/basicstation/files/7716085/debug-patch.zip)

beitler, 2022-01-31:
Most such issues arise from authentication failures. Please use the flag `-l AIO:XDEBUG` to inspect such scenarios. With that log level set, station will print more details on the message exchange with the backend. Setting the environment variable `STATION_TLSDBG=4` also helps to inspect the details of the TLS handshake.

closed 2022-01-31 by beitler