lorabasics/basicstation #119: seqno field in dntxed messages

RadW2020, 2021-08-17:
Hi, 

I get **_dntxed_** messages like this:

```
{"msgtype":"dntxed","seqno":0,"diid":0,"DR":5,"Freq":868100000,"DevEui":"68-49-B5-28-68-49-B5-28","rctx":0,"xtime":37436176804343530,"txtime":9079.010030,"gpstime":0}
```

But in the documentation, there is nothing regarding the seqno field.

[image]

is this ok?

cstratton, 2021-09-23:
If you look in the code, seqno is an old field having similar function to  the current diid

Typically a json parser simply ignores fields it doesn't know about, chances are the goal here is maximum compatibility so I wouldn't worry about it unless you are seeing some specific actual failure which you think might be traced back to this.

beitler, 2023-05-26:
> If you look in the code, seqno is an old field having similar function to the current diid
>
> Typically a json parser simply ignores fields it doesn't know about, chances are the goal here is maximum [...]

This is exactly right. The JSON message parser should ignore unknown/unspecified fields for forward and backward compatibility.

closed 2023-05-26 by beitler