lorabasics/basicstation #120: Concentrator start failed: lgw_start

Ryoyo123, 2021-08-30:
I would like your help to debug an issue.
 I have Semtech SX1302CxxxGW1 Corecell Gateway. I was setting up the Lora Basic station gateway in the SX1302CxxxGW1. I have implemented Lora Basic Station in the SX1302CxxxGW1 by following this guide  https://github.com/lorabasics/basicstation.

**Goal**
>   -Setup the Lora Basic Station in the SX1302CxxxGW1
>   -Connect AWS IoT Core for LoRaWAN network server

**Steps Executed:**
>    git clone https://github.com/lorabasics/basicstation.git
>    cd basicstation
>    make platform=corecell variant=std
>    RADIODEV=/dev/spidev0.0 /home/pi/basicstation/build-corecell-std/bin/station

 I have configured with all the necessary credential files( cups.uri, cups.key, cups.trust, cups-bak.uri, cups-bak.key, cups-bak.trust, cups-boot.uri, cups-boot.key, cups-boot.trust, tc.uri, tc.key, tc.trust, tc-bak.uri, tc-bak.key, tc-bak.trust) trying to connect to AWS IoT Core for LoRaWAN.

 When I running the station by using, 
 **RADIODEV=/dev/spidev0.0 /home/pi/basicstation/build-corecell-std/bin/station**
 I got the following errors in the gateway logs.

 SX1302 Corecell Gateway Logs
```
cert. version     : 3
serial number     : F5:05:F6:2B:9B:C7:EA:88:71:9F:40:CB:A1:DE:22:85:E8:8F:D0:39
issuer name       : OU=Amazon Web Services O=Amazon.com Inc. L=Seattle ST=Washington C=US
subject name      : CN=AWS IoT Certificate
issued  on        : 2021-08-25 11:40:57
expires on        : 2049-12-31 23:59:59
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic constraints : CA=false
key usage         : Digital Signature
2021-08-27 15:10:20.159 [AIO:INFO]
2021-08-27 15:10:20.390 [TCE:VERB] Connecting to MUXS...
2021-08-27 15:10:20.390 [AIO:DEBU] [4] HTTP connection shutdown...
2021-08-27 15:10:20.391 [CUP:INFO] Interaction with CUPS done (no updates) - next regular check in 1d
2021-08-27 15:10:21.175 [TCE:VERB] Connected to MUXS.
ERROR: Failed to stop TX trigger
ERROR: Failed to stop TX trigger
ERROR: failed to close I2C device (err=-1)
2021-08-27 15:10:21.369 [S2E:WARN] Unknown field in router_config - ignored: protocol (0xFD309030)
2021-08-27 15:10:21.369 [S2E:WARN] Unknown field in router_config - ignored: regionid (0xE6FFB211)
2021-08-27 15:10:21.372 [RAL:INFO] Lora gateway library version: Version: 1.0.5;
2021-08-27 15:10:21.387 [RAL:VERB] Connecting to device: /dev/spidev0.0
2021-08-27 15:10:21.387 [RAL:DEBU] SX130x txlut table (0 entries)
2021-08-27 15:10:21.387 [RAL:VERB] SX1302 rssi_tcomp: coeff_a=0.000 coeff_b=0.000 coeff_c=0.000 coeff_d=0.000 coeff_e=0.000
2021-08-27 15:10:21.387 [RAL:VERB] SX1302 rxrfchain 0: enable=1 freq=922.2MHz rssi_offset=-166.000000 type=2 tx_enable=1
2021-08-27 15:10:21.387 [RAL:VERB] SX1302 rxrfchain 1: enable=1 freq=923.2MHz rssi_offset=-166.000000 type=2 tx_enable=0
2021-08-27 15:10:21.387 [RAL:VERB] SX1302 ifchain  0: enable=1 rf_chain=0 freq=-200000 bw=0 SF=0 sync_word=0/0
2021-08-27 15:10:21.387 [RAL:VERB] SX1302 ifchain  1: enable=1 rf_chain=0 freq=0 bw=0 SF=0 sync_word=0/0
2021-08-27 15:10:21.387 [RAL:VERB] SX1302 ifchain  2: enable=1 rf_chain=0 freq=0 bw=0 SF=0 sync_word=0/0
2021-08-27 15:10:21.387 [RAL:VERB] SX1302 ifchain  3: enable=1 rf_chain=0 freq=400000 bw=0 SF=0 sync_word=0/0
2021-08-27 15:10:21.387 [RAL:VERB] SX1302 ifchain  4: enable=1 rf_chain=1 freq=-400000 bw=0 SF=0 sync_word=0/0
2021-08-27 15:10:21.387 [RAL:VERB] SX1302 ifchain  5: enable=1 rf_chain=1 freq=-200000 bw=0 SF=0 sync_word=0/0
2021-08-27 15:10:21.387 [RAL:VERB] SX1302 ifchain  6: enable=1 rf_chain=1 freq=0 bw=0 SF=0 sync_word=0/0
2021-08-27 15:10:21.387 [RAL:VERB] SX1302 ifchain  7: enable=1 rf_chain=1 freq=200000 bw=0 SF=0 sync_word=0/0
2021-08-27 15:10:21.387 [RAL:VERB] SX1302 ifchain  8: enable=1 rf_chain=0 freq=-100000 bw=5 SF=7 sync_word=0/0 [STD] Explicit header
2021-08-27 15:10:21.387 [RAL:VERB] SX1302 ifchain  9: enable=1 rf_chain=0 freq=0 bw=0 SF=0 sync_word=0/0
2021-08-27 15:10:21.387 [RAL:VERB] Station device: /dev/spidev0.0 (PPS capture disabled)
2021-08-27 15:10:21.395 [RAL:ERRO] Concentrator start failed: lgw_start
2021-08-27 15:10:21.395 [RAL:ERRO] ral_config failed with status 0x08
2021-08-27 15:10:21.395 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2021-08-27 15:10:21.395 [AIO:DEBU] [3] ws_close reason=1000
2021-08-27 15:10:21.396 [AIO:DEBU] Echoing close - reason=1000
2021-08-27 15:10:21.565 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2021-08-27 15:10:21.565 [AIO:DEBU] [3] WS connection shutdown...
2021-08-27 15:10:21.566 [TCE:VERB] Connection to MUXS closed in state -1
2021-08-27 15:10:21.566 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)
ERROR: Failed to stop TX trigger
ERROR: Failed to stop TX trigger
ERROR: failed to close I2C device (err=-1)
2021-08-27 15:10:31.569 [any:INFO] ./tc.trust:
cert. version     : 3
serial number     : A7:0E:4A:4C:34:82:B7:7F
```

Question
> 1)	Is it due to the SPI connection between the SX1302 board and the Raspberry Pi ?
> 2)	How can I start the Concentrator (Lora Basic Station in the SX1302CxxxGW1) ?

alifrugal, 2021-09-03:
You might need to reset the LoRa concentrator that could be in failed state. You can use this kind of script and adapt it to the pin used in your gateway to reset. 

```
#!/bin/bash

echo "Restarting the CE0 pin..."
CSPIN=8
echo "$CSPIN" > /sys/class/gpio/export
sleep 1
echo "out" > /sys/class/gpio/gpio$CSPIN/direction
echo "0" > /sys/class/gpio/gpio$CSPIN/value
sleep 1
echo "1" > /sys/class/gpio/gpio$CSPIN/value
echo "Done\n"
sleep 1

echo "Restarting the Concentrator..."
RSTPIN=17
echo "$RSTPIN" > /sys/class/gpio/export
sleep 1
echo "out" > /sys/class/gpio/gpio$RSTPIN/direction
echo "1" > /sys/class/gpio/gpio$RSTPIN/value
sleep 1
echo "0" > /sys/class/gpio/gpio$RSTPIN/value
echo "Done\n"
sleep 2
```

Ryoyo123, 2021-09-09:
Hi Ali-san,
Thank you for your advice. I created a file called "test.sh" using the script you suggested and executed it. However, I got the following error. The file permissions are set to be writable.
 What is the cause? 　And how should I change the settings?
```
pi@rasPi3-LoRa:~ $ ./test.sh
Restarting the CE0 pin...
./test.sh: line 5: echo: write error: Device or resource busy
./test.sh: line 7: /sys/class/gpio/gpio8/direction: No such file or directory
./test.sh: line 8: /sys/class/gpio/gpio8/value: No such file or directory
./test.sh: line 10: /sys/class/gpio/gpio8/value: No such file or directory
Done\n
Restarting the Concentrator...
./test.sh: line 16: echo: write error: Device or resource busy
Done\n
pi@rasPi3-LoRa:~ $
```

And also, is the reason why there is no "gpio8" in "/ sys / class / gpio /" ？　Because the line5 command is not executed?
```
i@rasPi3-LoRa:~ $ ls /sys/class/gpio/
export  gpio17  gpiochip0  gpiochip100  gpiochip504  unexport
pi@rasPi3-LoRa:~ $
```
Could you suggest me some kind advice?

alifrugal, 2021-09-09:
Could you try to kill existing basics station processes with this command: 
`kill -9 $(ps aux | grep '[s]tation' | awk '{print $2}')`

and then try again to launch basics station with this command (in the according folder): 
`RADIODEV=/dev/spidev0.0 ./station -f`

Ryoyo123, 2021-09-26:
Hi Ali-san,
 Thank you for your many advices.
 I'm reconfiguring, refer to your YouTube demo video. 
>   https://www.youtube.com/watch?v=6-ZrdRjqdTk

You are using RAK7246. And the commands introduced in Video are as follows. 

```
Tool
RAK7246

Command
$ sudo raspi-config
  [Interface Option]
    SPI enable
$sudo apt-get update
$sudo apt-get install git
$git clone https://github.com/lorabasics/basicstation.git
$cd basicstation/
$make platform=rpi variant=std
$cd build-rpi-std/bin/
$cd
$cp basicstation/examples/live-s2.sm.tc/station.conf station.conf

AWS management Console

$nano cups.uri
$nano cups.trust
$nano cups.crt
$nano cups.key

$RADIODEV=/dev/spidev0.0 /home/pi/basicstation/build-rpi-std/bin/station
```
**Question**

> I am using Raspberry Pi + SX1302 Corecell GW Concentrator. For Corecell, where and how should I change it?

alifrugal, 2021-09-27:
Hi, 
Do you have any issue using the same commands than in the video with your hardware?

Ryoyo123, 2021-09-28:
2021.09.28
Hi Ali-san,

Thank you for your advice.

The LoRa Gateway chip mounted on the RAK7246 is SX1308 (= SX1301). However, the LoRa gateway chip mounted on the Corecell is SX1302. SX1302 chip is a new version. 
I executed Ali's command as it is. Then, the program for SX1301 started as follows.
```
cert. version     : 3
serial number     : A7:0E:4A:4C:34:82:B7:7F
issuer name       : C=US, O=Starfield Technologies, Inc., OU=Starfield Class 2 Certification Authority
subject name      : C=US, ST=Arizona, L=Scottsdale, O=Starfield Technologies, Inc., CN=Starfield Services Root Certificate Authority - G2
issued  on        : 2009-09-02 00:00:00
expires on        : 2034-06-28 17:39:16
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic con2021-09-28 12:48:46.270 [any:INFO] ./tc.crt:
cert. version     : 3
serial number     : 6B:68:88:E1:49:EE:DF:F3:FE:F7:B4:23:69:48:B2:EB:CC:A0:F4:3D
issuer name       : OU=Amazon Web Services O=Amazon.com Inc. L=Seattle ST=Washington C=US
subject name      : CN=AWS IoT Certificate
issued  on        : 2021-09-26 09:43:27
expires on        : 2049-12-31 23:59:59
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic constraints : CA=false
key usage         : Digital Signature
2021-09-28 12:48:46.270 [AIO:INFO]
2021-09-28 12:48:46.483 [TCE:VERB] Connecting to MUXS...
2021-09-28 12:48:46.483 [AIO:DEBU] [4] HTTP connection shutdown...
2021-09-28 12:48:46.483 [CUP:INFO] Interaction with CUPS done (no updates) - next regular check in 1d
2021-09-28 12:48:47.303 [TCE:VERB] Connected to MUXS.
2021-09-28 12:48:47.489 [S2E:WARN] Unknown field in router_config - ignored: protocol (0xFD309030)
2021-09-28 12:48:47.489 [S2E:WARN] Unknown field in router_config - ignored: regionid (0xE6FFB211)
2021-09-28 12:48:47.489 [RAL:INFO] Lora gateway library version: Version: 5.0.1;
2021-09-28 12:48:47.501 [RAL:VERB] Connecting to device: /dev/spidev0.0
2021-09-28 12:48:47.501 [RAL:DEBU] SX130x txlut table (0 entries)
2021-09-28 12:48:47.501 [RAL:VERB] SX1301 rxrfchain 0: enable=1 freq=922.2MHz rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
2021-09-28 12:48:47.501 [RAL:VERB] SX1301 rxrfchain 1: enable=1 freq=923.2MHz rssi_offset=-166.000000 type=2 tx_enable=0 tx_notch_freq=0
2021-09-28 12:48:47.501 [RAL:VERB] SX1301 ifchain  0: enable=1 rf_chain=0 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2021-09-28 12:48:47.502 [RAL:VERB] SX1301 ifchain  1: enable=1 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2021-09-28 12:48:47.502 [RAL:VERB] SX1301 ifchain  2: enable=1 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2021-09-28 12:48:47.502 [RAL:VERB] SX1301 ifchain  3: enable=1 rf_chain=0 freq=400000 bandwidth=0 datarate=0 sync_word=0/0
2021-09-28 12:48:47.502 [RAL:VERB] SX1301 ifchain  4: enable=1 rf_chain=1 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2021-09-28 12:48:47.502 [RAL:VERB] SX1301 ifchain  5: enable=1 rf_chain=1 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2021-09-28 12:48:47.502 [RAL:VERB] SX1301 ifchain  6: enable=1 rf_chain=1 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2021-09-28 12:48:47.502 [RAL:VERB] SX1301 ifchain  7: enable=1 rf_chain=1 freq=200000 bandwidth=0 datarate=0 sync_word=0/0
2021-09-28 12:48:47.502 [RAL:VERB] SX1301 ifchain  8: enable=1 rf_chain=0 freq=-100000 bandwidth=2 datarate=2 sync_word=0/0
2021-09-28 12:48:47.502 [RAL:VERB] SX1301 ifchain  9: enable=1 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2021-09-28 12:48:47.502 [RAL:VERB] SX130x LBT not enabled
2021-09-28 12:48:47.502 [RAL:VERB] Station device: /dev/spidev0.0 (PPS capture disabled)
2021-09-28 12:48:47.504 [RAL:ERRO] Concentrator start failed: lgw_start
2021-09-28 12:48:47.504 [RAL:ERRO] ral_config failed with status 0x08
2021-09-28 12:48:47.504 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2021-09-28 12:48:47.504 [AIO:DEBU] [3] ws_close reason=1000
2021-09-28 12:48:47.504 [AIO:DEBU] Echoing close - reason=1000
2021-09-28 12:48:47.678 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2021-09-28 12:48:47.678 [AIO:DEBU] [3] WS connection shutdown...
2021-09-28 12:48:47.678 [TCE:VERB] Connection to MUXS closed in state -1
2021-09-28 12:48:47.678 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)
2021-09-28 12:48:57.680 [any:INFO] ./tc.trust:
```
Please check station.conf
```
{
    /* If slave-X.conf present this acts as default settings */
    "SX1301_conf": {                 /* Actual channel plan is controlled by server */
        "lorawan_public": true,      /* is default */
        "clksrc": 1,                 /* radio_1 provides clock to concentrator */
        /* path to the SPI device, un-comment if not specified on the command line e.g., RADIODEV=/dev/spidev0.0 */
[... 152 lines cut ...]
                {"rf_power": 27, "pa_gain": 1, "pwr_idx": 14}
            ]
        },
        "radio_1": {
            "type": "SX1250",
            "rssi_offset": -215.4,
            "rssi_tcomp": {"coeff_a": 0, "coeff_b": 0, "coeff_c": 20.41, "coeff_d": 2162.56, "coeff_e": 0},
            "tx_enable": false
        }
        /* chan_multiSF_X, chan_Lora_std, chan_FSK provided by LNS */
    },
    "station_conf": {
        "radio_init": "rinit.sh",
        "RADIO_INIT_WAIT": "5s",
        "RX_POLL_INTV": "10ms",
        "TC_TIMEOUT": "360s",
        "log_file":  "stderr",   /* "station.log" */
        "log_level": "DEBUG",  /* XDEBUG,DEBUG,VERBOSE,INFO,NOTICE,WARNING,ERROR,CRITICAL */
        "log_size":  10000000,
        "log_rotate":  3
    }
}
```

If you notice anything, please give me some advice.

alifrugal, 2021-09-30:
Have you used the reset script before running Basics station ? You might need to run it using sudo and with the PIN (try with PIN22 instead of PIN17) that is used in the LoRa concentrator to reset it. 

`#!/bin/bash

echo "Restarting the CE0 pin..."

echo "Restarting the Concentrator..."
RSTPIN=22
echo "$RSTPIN" > /sys/class/gpio/export
sleep 1
echo "out" > /sys/class/gpio/gpio$RSTPIN/direction
echo "1" > /sys/class/gpio/gpio$RSTPIN/value
sleep 1
echo "0" > /sys/class/gpio/gpio$RSTPIN/value
echo "Done\n"
sleep 2`

Ryoyo123, 2021-10-01:
Hi Ali-san,

Thank you for your advice.

The script rinit.sh consists of a series of commands that reset the wireless hardware. I copied this script rinit.sh into a field in station.conf. I also copied reset_lgw.sh used in it. I ran it and it succeeded.

 Then, the program for SX1302Corecell GW started as follows.
```
cert. version     : 3
serial number     : A7:0E:4A:4C:34:82:B7:7F
issuer name       : C=US, O=Starfield Technologies, Inc., OU=Starfield Class 2 Certification Authority
subject name      : C=US, ST=Arizona, L=Scottsdale, O=Starfield Technologies, Inc., CN=Starfield Services Root Certificate Authority - G2
issued  on        : 2009-09-02 00:00:00
expires on        : 2034-06-28 17:39:16
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic con2021-10-01 13:53:45.241 [any:INFO] ./tc.crt:
cert. version     : 3
serial number     : 27:40:E9:E7:2E:0F:95:1A:DC:68:B2:38:54:95:26:55:DD:DF:EC:9A
issuer name       : OU=Amazon Web Services O=Amazon.com Inc. L=Seattle ST=Washington C=US
subject name      : CN=AWS IoT Certificate
issued  on        : 2021-10-01 13:51:43
expires on        : 2049-12-31 23:59:59
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic constraints : CA=false
key usage         : Digital Signature
2021-10-01 13:53:45.241 [AIO:INFO]
2021-10-01 13:53:45.473 [TCE:VERB] Connecting to MUXS...
2021-10-01 13:53:46.343 [TCE:VERB] Connected to MUXS.
2021-10-01 13:53:46.544 [S2E:WARN] Unknown field in router_config - ignored: protocol (0xFD309030)
2021-10-01 13:53:46.544 [S2E:WARN] Unknown field in router_config - ignored: regionid (0xE6FFB211)
2021-10-01 13:53:46.545 [SYS:VERB] rinit.sh: Forked, waiting...
CoreCell reset through GPIO23...
CoreCell power enable through GPIO18...
2021-10-01 13:53:47.297 [SYS:INFO] Process rinit.sh (pid=5942) completed
ERROR: Failed to stop TX trigger
ERROR: Failed to stop TX trigger
ERROR: failed to close I2C device (err=-1)
2021-10-01 13:53:47.297 [RAL:INFO] Lora gateway library version: Version: 1.0.5;
2021-10-01 13:53:47.310 [RAL:VERB] Connecting to device: /dev/spidev0.0
2021-10-01 13:53:47.310 [RAL:DEBU] SX130x txlut table (16 entries)
2021-10-01 13:53:47.310 [RAL:VERB] SX1302 txlut  0:  rf_power=12 pa_gain=0 pwr_idx=15
2021-10-01 13:53:47.310 [RAL:VERB] SX1302 txlut  1:  rf_power=13 pa_gain=0 pwr_idx=16
2021-10-01 13:53:47.310 [RAL:VERB] SX1302 txlut  2:  rf_power=14 pa_gain=0 pwr_idx=17
2021-10-01 13:53:47.310 [RAL:VERB] SX1302 txlut  3:  rf_power=15 pa_gain=0 pwr_idx=19
2021-10-01 13:53:47.310 [RAL:VERB] SX1302 txlut  4:  rf_power=16 pa_gain=0 pwr_idx=20
2021-10-01 13:53:47.310 [RAL:VERB] SX1302 txlut  5:  rf_power=17 pa_gain=0 pwr_idx=22
2021-10-01 13:53:47.310 [RAL:VERB] SX1302 txlut  6:  rf_power=18 pa_gain=1 pwr_idx=1
2021-10-01 13:53:47.310 [RAL:VERB] SX1302 txlut  7:  rf_power=19 pa_gain=1 pwr_idx=2
2021-10-01 13:53:47.310 [RAL:VERB] SX1302 txlut  8:  rf_power=20 pa_gain=1 pwr_idx=3
2021-10-01 13:53:47.310 [RAL:VERB] SX1302 txlut  9:  rf_power=21 pa_gain=1 pwr_idx=4
2021-10-01 13:53:47.310 [RAL:VERB] SX1302 txlut 10:  rf_power=22 pa_gain=1 pwr_idx=5
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 txlut 11:  rf_power=23 pa_gain=1 pwr_idx=6
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 txlut 12:  rf_power=24 pa_gain=1 pwr_idx=7
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 txlut 13:  rf_power=25 pa_gain=1 pwr_idx=9
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 txlut 14:  rf_power=26 pa_gain=1 pwr_idx=11
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 txlut 15:  rf_power=27 pa_gain=1 pwr_idx=14
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 rssi_tcomp: coeff_a=0.000 coeff_b=0.000 coeff_c=20.410 coeff_d=2162.560 coeff_e=0.000
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 rxrfchain 0: enable=1 freq=922.2MHz rssi_offset=-215.399994 type=5 tx_enable=1
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 rxrfchain 1: enable=1 freq=923.2MHz rssi_offset=-215.399994 type=5 tx_enable=0
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 ifchain  0: enable=1 rf_chain=0 freq=-200000 bw=0 SF=0 sync_word=0/0
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 ifchain  1: enable=1 rf_chain=0 freq=0 bw=0 SF=0 sync_word=0/0
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 ifchain  2: enable=1 rf_chain=0 freq=0 bw=0 SF=0 sync_word=0/0
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 ifchain  3: enable=1 rf_chain=0 freq=400000 bw=0 SF=0 sync_word=0/0
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 ifchain  4: enable=1 rf_chain=1 freq=-400000 bw=0 SF=0 sync_word=0/0
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 ifchain  5: enable=1 rf_chain=1 freq=-200000 bw=0 SF=0 sync_word=0/0
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 ifchain  6: enable=1 rf_chain=1 freq=0 bw=0 SF=0 sync_word=0/0
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 ifchain  7: enable=1 rf_chain=1 freq=200000 bw=0 SF=0 sync_word=0/0
2021-10-01 13:53:47.311 [RAL:VERB] SX1302 ifchain  8: enable=1 rf_chain=0 freq=-100000 bw=5 SF=7 sync_word=0/0 [STD] Explicit header
[... 4 lines cut ...]
2021-10-01 13:53:49.692 [S2E:VERB]   DR0  SF12/BW125
2021-10-01 13:53:49.692 [S2E:VERB]   DR1  SF11/BW125
2021-10-01 13:53:49.692 [S2E:VERB]   DR2  SF10/BW125
2021-10-01 13:53:49.692 [S2E:VERB]   DR3  SF9/BW125
2021-10-01 13:53:49.692 [S2E:VERB]   DR4  SF8/BW125
2021-10-01 13:53:49.692 [S2E:VERB]   DR5  SF7/BW125
2021-10-01 13:53:49.692 [S2E:VERB]   DR6  SF7/BW250
2021-10-01 13:53:49.692 [S2E:VERB]   DR7  FSK
2021-10-01 13:53:49.692 [S2E:VERB]   DR8  undefined
2021-10-01 13:53:49.692 [S2E:VERB]   DR9  undefined
2021-10-01 13:53:49.692 [S2E:VERB]   DR10 undefined
[... cut ...]
2021-10-01 13:53:49.693 [S2E:VERB]   Dev/test settings: nocca=0 nodc=0 nodwell=0
2021-10-01 13:54:31.699 [SYN:INFO] MCU/SX130X drift stats: min: +0.0ppm  q50: +11.9ppm  q80: +21.4ppm  max: +35.2ppm - threshold q90: +24.8ppm
2021-10-01 13:54:31.699 [SYN:INFO] Mean MCU drift vs SX130X#0: 11.9ppm
2021-10-01 13:54:48.501 [SYN:INFO] Time sync qualities: min=132 q90=258 max=267 (previous q90=2147483647)
2021-10-01 13:55:10.554 [SYN:INFO] MCU/SX130X drift stats: min: -1.9ppm  q50: +9.5ppm  q80: +21.4ppm  max: +29.5ppm - threshold q90: +27.6ppm
```

alifrugal, 2021-10-04:
Looking at the log, it seems that your gateway is now successfully connected to the LNS.

TokenZero, 2021-10-05:
Hi, 

I'm having a similar issue and I've followed @Ryoyo123 steps to the letter using corecell as the platform and copying the two reset .sh files. The only difference it I'm using a RAK2287 concentrator, so I've altered the reset pin in reset_lgw.sh (changed to 17) to account for the different hardware connections. I left the power pin as 18 since it is unconnected on the RAK2287. The output I'm getting is below.

`
pi@raspberrypi:~/basicstation $ RADIODEV=/dev/spidev0.0 build-corecell-std/bin/station
2021-10-05 09:54:03.572 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotate=3)
2021-10-05 09:54:03.572 [SYS:INFO] Station Ver : 2.0.5(corecell/std) 2021-10-05 09:44:17
2021-10-05 09:54:03.572 [SYS:INFO] Package Ver : (null)
2021-10-05 09:54:03.572 [SYS:INFO] proto EUI   : 0:dca6:3205:f1a        (/sys/class/net/eth0/address)
2021-10-05 09:54:03.572 [SYS:INFO] prefix EUI  : ::1    (builtin)
2021-10-05 09:54:03.572 [SYS:INFO] Station EUI : dca6:32ff:fe05:f1a
2021-10-05 09:54:03.572 [SYS:INFO] Station home: ./     (builtin)
2021-10-05 09:54:03.572 [SYS:INFO] Station temp: /var/tmp/      (builtin)
2021-10-05 09:54:03.774 [TCE:INFO] Starting TC engine
2021-10-05 09:54:03.775 [any:INFO] ./tc.trust:
cert. version     : 3
serial number     : A7:0E:4A:4C:34:82:B7:7F
issuer name       : C=US, O=Starfield Technologies, Inc., OU=Starfield Class 2 Certification Authority
subject name      : C=US, ST=Arizona, L=Scottsdale, O=Starfield Technologies, Inc., CN=Starfield Services Root Certificate Authority - G2
issued  on        : 2009-09-02 00:00:00
expires on        : 2034-06-28 17:39:16
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic con2021-10-05 09:54:03.781 [any:INFO] ./tc.crt:
cert. version     : 3
serial number     : 5D:D1:7B:E4:F2:3F:36:85:54:FA:4B:4E:55:5E:1A:CE:D2:2D:87:7B
issuer name       : OU=Amazon Web Services O=Amazon.com Inc. L=Seattle ST=Washington C=US
subject name      : CN=AWS IoT Certificate
issued  on        : 2021-10-05 09:47:42
expires on        : 2049-12-31 23:59:59
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic constraints : CA=false
key usage         : Digital Signature
2021-10-05 09:54:03.781 [AIO:INFO]
2021-10-05 09:54:04.014 [TCE:INFO] Connecting to INFOS: wss://A3OQ74V8656RDU.lns.lorawan.us-west-2.amazonaws.com:443
2021-10-05 09:54:04.014 [CUP:INFO] Starting a CUPS session in 0 seconds.
2021-10-05 09:54:04.014 [CUP:INFO] Starting a CUPS session now.
2021-10-05 09:54:04.014 [CUP:INFO] Connecting to CUPS ... https://A3OQ74V8656RDU.cups.lorawan.us-west-2.amazonaws.com:443 (try #1)
2021-10-05 09:54:04.014 [any:INFO] ./cups.trust:
cert. version     : 3
serial number     : A7:0E:4A:4C:34:82:B7:7F
issuer name       : C=US, O=Starfield Technologies, Inc., OU=Starfield Class 2 Certification Authority
subject name      : C=US, ST=Arizona, L=Scottsdale, O=Starfield Technologies, Inc., CN=Starfield Services Root Certificate Authority - G2
issued  on        : 2009-09-02 00:00:00
expires on        : 2034-06-28 17:39:16
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic c2021-10-05 09:54:04.022 [any:INFO] ./cups.crt:
cert. version     : 3
serial number     : 2F:B2:92:89:7F:3C:5C:D7:89:CB:36:E6:5A:1B:82:93:A7:9D:AF:B6
issuer name       : OU=Amazon Web Services O=Amazon.com Inc. L=Seattle ST=Washington C=US
subject name      : CN=AWS IoT Certificate
issued  on        : 2021-09-27 10:25:44
expires on        : 2049-12-31 23:59:59
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic constraints : CA=false
key usage         : Digital Signature
2021-10-05 09:54:04.022 [AIO:INFO]
2021-10-05 09:54:04.266 [CUP:VERB] Retrieving update-info from CUPS https://A3OQ74V8656RDU.cups.lorawan.us-west-2.amazonaws.com:443...
2021-10-05 09:54:05.057 [AIO:DEBU] [4] HTTP connection shutdown...
2021-10-05 09:54:05.057 [CUP:INFO] Interaction with CUPS done (no updates) - next regular check in 1d
2021-10-05 09:54:05.108 [TCE:INFO] Infos: dca6:32ff:fe05:0f1a dca6:32ff:fe05:0f1a wss://A3OQ74V8656RDU.lns.lorawan.us-west-2.amazonaws.com:443/gateway/dca632fffe050f1a
2021-10-05 09:54:05.108 [AIO:DEBU] [3] ws_close reason=1000
2021-10-05 09:54:05.108 [AIO:DEBU] Echoing close - reason=1000
2021-10-05 09:54:05.108 [AIO:DEBU] [3|WS] Server sent close: reason=1006
2021-10-05 09:54:05.108 [AIO:DEBU] [3] WS connection shutdown...
2021-10-05 09:54:05.108 [any:INFO] ./tc.trust:
[... 52 lines cut ...]
2021-10-05 09:54:07.129 [RAL:VERB] SX1302 rxrfchain 1: enable=1 freq=868.5MHz rssi_offset=-215.399994 type=5 tx_enable=0
2021-10-05 09:54:07.129 [RAL:VERB] SX1302 ifchain  0: enable=1 rf_chain=0 freq=-400000 bw=0 SF=0 sync_word=0/0
2021-10-05 09:54:07.129 [RAL:VERB] SX1302 ifchain  1: enable=1 rf_chain=0 freq=-200000 bw=0 SF=0 sync_word=0/0
2021-10-05 09:54:07.129 [RAL:VERB] SX1302 ifchain  2: enable=1 rf_chain=0 freq=0 bw=0 SF=0 sync_word=0/0
2021-10-05 09:54:07.129 [RAL:VERB] SX1302 ifchain  3: enable=1 rf_chain=0 freq=200000 bw=0 SF=0 sync_word=0/0
2021-10-05 09:54:07.129 [RAL:VERB] SX1302 ifchain  4: enable=1 rf_chain=0 freq=400000 bw=0 SF=0 sync_word=0/0
2021-10-05 09:54:07.129 [RAL:VERB] SX1302 ifchain  5: enable=1 rf_chain=1 freq=-400000 bw=0 SF=0 sync_word=0/0
2021-10-05 09:54:07.129 [RAL:VERB] SX1302 ifchain  6: enable=1 rf_chain=1 freq=-200000 bw=0 SF=0 sync_word=0/0
2021-10-05 09:54:07.129 [RAL:VERB] SX1302 ifchain  7: enable=1 rf_chain=1 freq=0 bw=0 SF=0 sync_word=0/0
[... cut ...]
2021-10-05 09:54:09.669 [RAL:ERRO] ral_config failed with status 0x08
2021-10-05 09:54:09.669 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2021-10-05 09:54:09.669 [AIO:DEBU] [3] ws_close reason=1000
2021-10-05 09:54:09.669 [AIO:DEBU] Echoing close - reason=1000
2021-10-05 09:54:09.853 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2021-10-05 09:54:09.853 [AIO:DEBU] [3] WS connection shutdown...
2021-10-05 09:54:09.854 [TCE:VERB] Connection to MUXS closed in state -1
2021-10-05 09:54:09.854 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)
`

chaudhariatul, 2022-10-15:
@Ryoyo123  Kindly share your configuration details... did you use corecell config and then the station.conf with rinit and reset_lgw?