lorabasics/basicstation #121: USB based SX1303

nmaas87, 2021-09-03:
Any thoughts on supporting SX1303 with USB?
I want to use the RAK5146 with SX1303, LBT and GPS - which only comes with USB (SPI version does not support Listen Before Talk, probably because of being too bottlenecked by SPI)

https://store.rakwireless.com/products/wislink-lpwan-concentrator-rak5146?variant=39677269409990

Thanks a lot :)

https://github.com/lorabasics/basicstation/issues/89

tonysmith55, 2021-09-03:
"probably because of being too bottlenecked by SPI"
If you look at the RAK datasheet and circuit schematic you will find to achieve LBT requires a third receiver chip.  The USB version of the RAK 5146 has a USB to SPI convertor to connect to the SX1303 chip.  The same convertor has a second SPI port to connect to the third receiver which performs LBT.  It's not about the performance of SPI vs USB but simply the circuit schematic adopted by RAK.

nmaas87, 2021-09-03:
Thanks @tonysmith55 for the insight, but I see no SPI version with LBT for sale.
However, this was a bit off-topic from me, sorry, I just wanted to ask to include USB based SX1303 support into the basicstation :).
Thank you :)

tonysmith55, 2021-09-03:
While I did not fully explain, given the RAK design the only way that LBT can be achieved is via USB.   This gives additional support for your request.

closed 2022-01-31 by beitler