lorabasics/basicstation #125: ERROR: failed to configure the temperature sensor

TokenZero, 2021-10-05:
I am attempting to get this running using a RPi 4 and a RAK2287 on AWS with CUPS. I keep getting the error `Concentrator start failed: lgw_start` which appears to be as a result of the temperature sensor failure. 

I found[ this page](https://github.com/JoToSystems/SX1302_UDP_Packet_Forwarder) which describes modification to the gateway_conf.json file to tell the packet forwarder to ignore the temperature sensor, however I'm not sure I'm able to find the correct place to put it.

As instructed, I went in and added the following two lines...
```
"temperature_sensor": true,
"temperature_value":25.0
```
under the `"gateway_conf"` section of the basicstation files for my region:
 `/deps/lgw1302/platform-corecell/packet_forwarder/global_conf.json.sx1257.EU868`
and 
`/deps/lgw1302/git-repo/packet_forwarder/global_conf.json.sx1257.EU868`. 

These were the closest thing I have that resemble the global_conf.json file for the sx1302.
Follow this I did a `make clean` and `make platform=corecell variant=std`. Unfortunately this produced the same outcome as below.

`
pi@raspberrypi:~/basicstation $ RADIODEV=/dev/spidev0.0 build-corecell-std/bin/station
2021-10-05 11:51:18.114 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotate=3)
2021-10-05 11:51:18.115 [SYS:INFO] Station Ver : 2.0.5(corecell/std) 2021-10-05 11:47:18
2021-10-05 11:51:18.115 [SYS:INFO] Package Ver : (null)
2021-10-05 11:51:18.115 [SYS:INFO] proto EUI   : 0:dca6:3205:f1a        (/sys/class/net/eth0/address)
2021-10-05 11:51:18.115 [SYS:INFO] prefix EUI  : ::1    (builtin)
2021-10-05 11:51:18.115 [SYS:INFO] Station EUI : dca6:32ff:fe05:f1a
2021-10-05 11:51:18.116 [SYS:INFO] Station home: ./     (builtin)
2021-10-05 11:51:18.116 [SYS:INFO] Station temp: /var/tmp/      (builtin)
2021-10-05 11:51:18.317 [TCE:INFO] Starting TC engine
2021-10-05 11:51:18.318 [any:INFO] ./tc.trust:
cert. version     : 3
serial number     : A7:0E:4A:4C:34:82:B7:7F
issuer name       : C=US, O=Starfield Technologies, Inc., OU=Starfield Class 2 Certification Authority
subject name      : C=US, ST=Arizona, L=Scottsdale, O=Starfield Technologies, Inc., CN=Starfield Services Root Certificate Authority - G2
issued  on        : 2009-09-02 00:00:00
expires on        : 2034-06-28 17:39:16
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic con2021-10-05 11:51:18.325 [any:INFO] ./tc.crt:
cert. version     : 3
serial number     : 5D:D1:7B:E4:F2:3F:36:85:54:FA:4B:4E:55:5E:1A:CE:D2:2D:87:7B
issuer name       : OU=Amazon Web Services O=Amazon.com Inc. L=Seattle ST=Washington C=US
subject name      : CN=AWS IoT Certificate
issued  on        : 2021-10-05 09:47:42
expires on        : 2049-12-31 23:59:59
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic constraints : CA=false
key usage         : Digital Signature
2021-10-05 11:51:18.325 [AIO:INFO]
2021-10-05 11:51:18.543 [TCE:INFO] Connecting to INFOS: wss://A3OQ74V8656RDU.lns.lorawan.us-west-2.amazonaws.com:443
2021-10-05 11:51:18.543 [CUP:INFO] Starting a CUPS session in 0 seconds.
2021-10-05 11:51:18.543 [CUP:INFO] Starting a CUPS session now.
2021-10-05 11:51:18.543 [CUP:INFO] Connecting to CUPS ... https://A3OQ74V8656RDU.cups.lorawan.us-west-2.amazonaws.com:443 (try #1)
2021-10-05 11:51:18.543 [any:INFO] ./cups.trust:
cert. version     : 3
serial number     : A7:0E:4A:4C:34:82:B7:7F
issuer name       : C=US, O=Starfield Technologies, Inc., OU=Starfield Class 2 Certification Authority
subject name      : C=US, ST=Arizona, L=Scottsdale, O=Starfield Technologies, Inc., CN=Starfield Services Root Certificate Authority - G2
issued  on        : 2009-09-02 00:00:00
expires on        : 2034-06-28 17:39:16
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic c2021-10-05 11:51:18.551 [any:INFO] ./cups.crt:
cert. version     : 3
serial number     : 2F:B2:92:89:7F:3C:5C:D7:89:CB:36:E6:5A:1B:82:93:A7:9D:AF:B6
issuer name       : OU=Amazon Web Services O=Amazon.com Inc. L=Seattle ST=Washington C=US
subject name      : CN=AWS IoT Certificate
issued  on        : 2021-09-27 10:25:44
expires on        : 2049-12-31 23:59:59
signed using      : RSA with SHA-256
[... 65 lines cut ...]
2021-10-05 11:51:21.644 [RAL:VERB] SX1302 rxrfchain 0: enable=1 freq=867.5MHz rssi_offset=-215.399994 type=5 tx_enable=1
2021-10-05 11:51:21.644 [RAL:VERB] SX1302 rxrfchain 1: enable=1 freq=868.5MHz rssi_offset=-215.399994 type=5 tx_enable=0
2021-10-05 11:51:21.644 [RAL:VERB] SX1302 ifchain  0: enable=1 rf_chain=0 freq=-400000 bw=0 SF=0 sync_word=0/0
2021-10-05 11:51:21.644 [RAL:VERB] SX1302 ifchain  1: enable=1 rf_chain=0 freq=-200000 bw=0 SF=0 sync_word=0/0
2021-10-05 11:51:21.644 [RAL:VERB] SX1302 ifchain  2: enable=1 rf_chain=0 freq=0 bw=0 SF=0 sync_word=0/0
2021-10-05 11:51:21.644 [RAL:VERB] SX1302 ifchain  3: enable=1 rf_chain=0 freq=200000 bw=0 SF=0 sync_word=0/0
2021-10-05 11:51:21.644 [RAL:VERB] SX1302 ifchain  4: enable=1 rf_chain=0 freq=400000 bw=0 SF=0 sync_word=0/0
2021-10-05 11:51:21.644 [RAL:VERB] SX1302 ifchain  5: enable=1 rf_chain=1 freq=-400000 bw=0 SF=0 sync_word=0/0
2021-10-05 11:51:21.644 [RAL:VERB] SX1302 ifchain  6: enable=1 rf_chain=1 freq=-200000 bw=0 SF=0 sync_word=0/0
2021-10-05 11:51:21.644 [RAL:VERB] SX1302 ifchain  7: enable=1 rf_chain=1 freq=0 bw=0 SF=0 sync_word=0/0
2021-10-05 11:51:21.644 [RAL:VERB] SX1302 ifchain  8: enable=1 rf_chain=1 freq=-200000 bw=5 SF=7 sync_word=0/0 [STD] Explicit header
2021-10-05 11:51:21.644 [RAL:VERB] SX1302 ifchain  9: enable=1 rf_chain=0 freq=0 bw=0 SF=0 sync_word=0/0
2021-10-05 11:51:21.644 [RAL:VERB] Station device: /dev/spidev0.0 (PPS capture disabled)
ERROR: failed to configure the temperature sensor
2021-10-05 11:51:24.186 [RAL:ERRO] Concentrator start failed: lgw_start
2021-10-05 11:51:24.186 [RAL:ERRO] ral_config failed with status 0x08
2021-10-05 11:51:24.186 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2021-10-05 11:51:24.186 [AIO:DEBU] [3] ws_close reason=1000
2021-10-05 11:51:24.186 [AIO:DEBU] Echoing close - reason=1000
2021-10-05 11:51:24.443 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2021-10-05 11:51:24.443 [AIO:DEBU] [3] WS connection shutdown...
2021-10-05 11:51:24.443 [TCE:VERB] Connection to MUXS closed in state -1
2021-10-05 11:51:24.443 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)

`

tonysmith55, 2021-10-05:
Something else you could look at
https://www.thethingsnetwork.org/forum/t/rp3-x-ic880a-gateway-stopped-working/15469/12

Not sure if this applies to a Pi4 as its using a different ARM chip to the Pi3 series, but may be worth having a look at.

TokenZero, 2021-10-05:
Hi Tony, thanks for your reply. 

The only software RAK seem to have is their [rak-common-for-gateway](https://github.com/RAKWireless/rak_common_for_gateway) which I'm not quite sure how to use to resolve this issue. They don't have any official examples of basicstation usage and rak-common-for-gateway doesn't work with AWS IoTCore, which I require.

Have you got any suggestions as to how I might make some modifications to get around this temperature sensor error?

The Pi4 is used because we're doing other processing on the edge with the gateway, so not totally wasted power :)
Don't think that other post is too relevant, unfortunately.

TokenZero, 2021-10-05:
I found a solution.

I basically had to do a `make platform=corecell variant` to pull in the dependencies (mainly https://github.com/Lora-net/sx1302_hal).

After that I edited `~/basicstation/deps/lgw1302/platform-corecell/libloragw/src/loragw_stts751.c` to comment out any I2C calls, and return a static temperature value as below;

`
/*
 / _____)             _              | |
( (____  _____ ____ _| |_ _____  ____| |__
 \____ \| ___ |    (_   _) ___ |/ ___)  _ \
 _____) ) ____| | | || |_| ____( (___| | | |
(______/|_____)_|_|_| \__)_____)\____)_| |_|
  (C)2019 Semtech

Description:
    Basic driver for ST ts751 temperature sensor

License: Revised BSD License, see LICENSE.TXT file include in the project
*/

/* -------------------------------------------------------------------------- */
/* --- DEPENDANCIES --------------------------------------------------------- */

#include <stdint.h>     /* C99 types */
#include <stdbool.h>    /* bool type */
#include <stdio.h>      /* printf fprintf */

#include "loragw_i2c.h"
#include "loragw_stts751.h"

/* -------------------------------------------------------------------------- */
/* --- PRIVATE MACROS ------------------------------------------------------- */

#define ARRAY_SIZE(a) (sizeof(a) / sizeof((a)[0]))
#if DEBUG_I2C == 1
    #define DEBUG_MSG(str)              fprintf(stdout, str)
    #define DEBUG_PRINTF(fmt, args...)  fprintf(stdout,"%s:%d: "fmt, __FUNCTION__, __LINE__, args)
    #define CHECK_NULL(a)               if(a==NULL){fprintf(stderr,"%s:%d: ERROR: NULL POINTER AS ARGUMENT\n", __FUNCTION__, __LINE__);return LGW_REG_ERROR;}
#else
    #define DEBUG_MSG(str)
    #define DEBUG_PRINTF(fmt, args...)
    #define CHECK_NULL(a)               if(a==NULL){return LGW_REG_ERROR;}
#endif

/* -------------------------------------------------------------------------- */
/* --- PRIVATE CONSTANTS ---------------------------------------------------- */

#define STTS751_REG_TEMP_H      0x00
#define STTS751_REG_STATUS      0x01
#define STTS751_STATUS_TRIPT    BIT(0)
#define STTS751_STATUS_TRIPL    BIT(5)
#define STTS751_STATUS_TRIPH    BIT(6)
#define STTS751_REG_TEMP_L      0x02
#define STTS751_REG_CONF        0x03
#define STTS751_CONF_RES_MASK   0x0C
#define STTS751_CONF_RES_SHIFT  2
#define STTS751_CONF_EVENT_DIS  BIT(7)
#define STTS751_CONF_STOP       BIT(6)
#define STTS751_REG_RATE        0x04
#define STTS751_REG_HLIM_H      0x05
#define STTS751_REG_HLIM_L      0x06
#define STTS751_REG_LLIM_H      0x07
#define STTS751_REG_LLIM_L      0x08
#define STTS751_REG_TLIM        0x20
#define STTS751_REG_HYST        0x21
#define STTS751_REG_SMBUS_TO    0x22

#define STTS751_REG_PROD_ID     0xFD
#define STTS751_REG_MAN_ID      0xFE
[... 100 lines cut ...]
    err = i2c_linuxdev_read(i2c_fd, i2c_addr, STTS751_REG_TEMP_L, &low_byte);
    if (err != 0) {
        printf("ERROR: failed to read I2C device 0x%02X (err=%i)\n", i2c_addr, err);
        return LGW_I2C_ERROR;
    }

    /* Read Temperature MSB */
    err = i2c_linuxdev_read(i2c_fd, i2c_addr, STTS751_REG_TEMP_H, &high_byte);
    if (err != 0) {
        printf("ERROR: failed to read I2C device 0x%02X (err=%i)\n", i2c_addr, err);
        return LGW_I2C_ERROR;
    }

    h = (int8_t)high_byte;
    *temperature = ((h << 8) | low_byte) / 256.0;
#endif
    *temperature = 25.0;

    DEBUG_PRINTF("Temperature: %f C (h:0x%02X l:0x%02X)\n", *temperature, 0, 0);

    return LGW_I2C_SUCCESS;
}

/* --- EOF ------------------------------------------------------------------ */
`

closed 2021-10-05 by TokenZero

chaudhariatul, 2022-11-28:
I have similar setup : RPi 4B + RAK2287 with 64bit OS. 

This thread helped resolve the 64bit OS issue: https://github.com/lorabasics/basicstation/issues/171#issuecomment-1263168451

---
**This is the error:**
```
CoreCell reset through GPIO17...
CoreCell power enable through GPIO18...
2022-11-28 05:41:17.892 [SYS:INFO] Process rinit.sh (pid=2725) completed
2022-11-28 05:41:17.892 [RAL:INFO] Lora gateway library version: Version: 2.1.0;
2022-11-28 05:41:17.895 [RAL:INFO] [LGW sx1302] full_duplex=0 clksrc=0 lorawan_public=1
2022-11-28 05:41:17.895 [RAL:DEBU] SX130x txlut table (16 entries)
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut  0:  rf_power=12 pa_gain=0 pwr_idx=15
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut  1:  rf_power=13 pa_gain=0 pwr_idx=16
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut  2:  rf_power=14 pa_gain=0 pwr_idx=17
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut  3:  rf_power=15 pa_gain=0 pwr_idx=19
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut  4:  rf_power=16 pa_gain=0 pwr_idx=20
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut  5:  rf_power=17 pa_gain=0 pwr_idx=22
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut  6:  rf_power=18 pa_gain=1 pwr_idx=1
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut  7:  rf_power=19 pa_gain=1 pwr_idx=2
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut  8:  rf_power=20 pa_gain=1 pwr_idx=3
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut  9:  rf_power=21 pa_gain=1 pwr_idx=4
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut 10:  rf_power=22 pa_gain=1 pwr_idx=5
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut 11:  rf_power=23 pa_gain=1 pwr_idx=6
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut 12:  rf_power=24 pa_gain=1 pwr_idx=7
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut 13:  rf_power=25 pa_gain=1 pwr_idx=9
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut 14:  rf_power=26 pa_gain=1 pwr_idx=11
2022-11-28 05:41:17.895 [RAL:INFO] SX1302 txlut 15:  rf_power=27 pa_gain=1 pwr_idx=14
2022-11-28 05:41:17.895 [RAL:INFO]  RX/TX RF0:    904.3MHz rssi_offset=-215.4 type=5 rssi_tcomp=0.000 0.000 20.410 2162.560 0.000
2022-11-28 05:41:17.895 [RAL:INFO]  RX    RF1:    905.3MHz rssi_offset=-215.4 type=5 rssi_tcomp=0.000 0.000 20.410 2162.560 0.000
2022-11-28 05:41:17.895 [RAL:INFO]  [mSF]   0:    903.9MHz rf=0 freq=-400.0 datarate=0
2022-11-28 05:41:17.895 [RAL:INFO]  [mSF]   1:    904.1MHz rf=0 freq=-200.0 datarate=0
2022-11-28 05:41:17.895 [RAL:INFO]  [mSF]   2:    904.3MHz rf=0 freq=  +0.0 datarate=0
2022-11-28 05:41:17.895 [RAL:INFO]  [mSF]   3:    904.5MHz rf=0 freq=+200.0 datarate=0
2022-11-28 05:41:17.895 [RAL:INFO]  [mSF]   4:    904.7MHz rf=0 freq=+400.0 datarate=0
2022-11-28 05:41:17.895 [RAL:INFO]  [mSF]   5:    904.9MHz rf=1 freq=-400.0 datarate=0
2022-11-28 05:41:17.895 [RAL:INFO]  [mSF]   6:    905.1MHz rf=1 freq=-200.0 datarate=0
2022-11-28 05:41:17.895 [RAL:INFO]  [mSF]   7:    905.3MHz rf=1 freq=  +0.0 datarate=0
2022-11-28 05:41:17.895 [RAL:INFO]  channel 8 disabled
2022-11-28 05:41:17.895 [RAL:INFO]  channel 9 disabled
2022-11-28 05:41:17.895 [RAL:INFO] Station device: spi:/dev/spidev0.0 (PPS capture disabled)
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
2022-11-28 05:41:17.895 [HAL:INFO] [lgw_com_open:84] Opening SPI communication interface
2022-11-28 05:41:17.895 [HAL:INFO] [lgw_spi_open:88] Setting SPI speed to 2000000
2022-11-28 05:41:17.895 [HAL:INFO] [lgw_connect:1192] chip version is 0x05 (v0.5)
2022-11-28 05:41:19.497 [HAL:ERRO] [lgw_start:906] failed to setup radio 0
2022-11-28 05:41:19.497 [RAL:ERRO] Concentrator start failed: lgw_start
2022-11-28 05:41:19.497 [RAL:ERRO] ral_config failed with status 0x08
2022-11-28 05:41:19.497 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2022-11-28 05:41:19.497 [AIO:DEBU] [3] ws_close reason=1000
2022-11-28 05:41:19.497 [AIO:DEBU] Echoing close - reason=1000
2022-11-28 05:41:19.539 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2022-11-28 05:41:19.539 [AIO:DEBU] [3] WS connection shutdown...
2022-11-28 05:41:19.539 [TCE:VERB] Connection to MUXS closed in state -1
2022-11-28 05:41:19.539 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)
2022-11-28 05:41:29.540 [AIO:INFO] ./tc.trust: 
```