lorabasics/basicstation #130: SX1302CSSxxxGW1 Concentrator start failed

Ryoyo123, 2021-11-01:
Hi , thank you for your advice.

Please help me to debug the problem again.
I have a Semtech **SX1302CSSxxxGW1** Corecell Gateway. It is a product with the **added LBT function.**
	
I was setting up a Lora Basic station gateway on the SX1302 CSSxxxGW1. I implemented Lora Basic Station on SX1302CSSxxxGW1 according to this guide https://github.com/lorabasics/basicstation.

Here, rinit.sh and reset_lgw.sh referenced in station.conf were also copied to the same level.

However, the Concentrator start failed.

I executed the command as follows:
```
Command <corecell>
$ sudo raspi-config
  [Interface Option]
    SPI enable
$sudo apt-get update
$sudo apt-get install git
$git clone https://github.com/lorabasics/basicstation.git
$cd basicstation/
$make platform=corecell variant=std
$cd build-corecell-std/bin/
$cd ~
$cp basicstation/examples/corecell/lns-ttn/station.conf station.conf
$cp basicstation/examples/corecell/rinit.sh rinit.sh
$cp basicstation/examples/corecell/reset_lgw.sh reset_lgw.sh

AWS management Console

$nano cups.uri
$nano cups.trust
$nano cups.crt
$nano cups.key
$RADIODEV=/dev/spidev0.0 /home/pi/basicstation/build-corecell-std/bin/station
```

But it doesn't start well.
```
cert. version     : 3
serial number     : 08:65:AE:7C:ED:76:FA:06:8C:E4:8E:4E:30:2D:EC:C6:A1:6A:C8:91
issuer name       : OU=Amazon Web Services O=Amazon.com Inc. L=Seattle ST=Washington C=US
subject name      : CN=AWS IoT Certificate
issued  on        : 2021-10-28 12:29:31
expires on        : 2049-12-31 23:59:59
signed using      : RSA with SHA-256
RSA key size      : 2048 bits
basic constraints : CA=false
key usage         : Digital Signature
2021-11-01 12:46:05.110 [AIO:INFO]
2021-11-01 12:46:05.395 [TCE:VERB] Connecting to MUXS...
2021-11-01 12:46:06.448 [TCE:VERB] Connected to MUXS.
2021-11-01 12:46:06.685 [S2E:WARN] Unknown field in router_config - ignored: protocol (0xFD309030)
2021-11-01 12:46:06.685 [S2E:WARN] Unknown field in router_config - ignored: regionid (0xE6FFB211)
2021-11-01 12:46:06.687 [SYS:VERB] rinit.sh: Forked, waiting...
CoreCell reset through GPIO23...
CoreCell power enable through GPIO18...
2021-11-01 12:46:07.650 [SYS:INFO] Process rinit.sh (pid=22195) completed
ERROR: Failed to stop TX trigger
ERROR: Failed to stop TX trigger
ERROR: failed to close I2C device (err=-1)
2021-11-01 12:46:07.650 [RAL:INFO] Lora gateway library version: Version: 1.0.5;
2021-11-01 12:46:07.655 [RAL:VERB] Connecting to device: /dev/spidev0.0
2021-11-01 12:46:07.655 [RAL:DEBU] SX130x txlut table (16 entries)
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut  0:  rf_power=12 pa_gain=0 pwr_idx=15
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut  1:  rf_power=13 pa_gain=0 pwr_idx=16
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut  2:  rf_power=14 pa_gain=0 pwr_idx=17
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut  3:  rf_power=15 pa_gain=0 pwr_idx=19
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut  4:  rf_power=16 pa_gain=0 pwr_idx=20
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut  5:  rf_power=17 pa_gain=0 pwr_idx=22
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut  6:  rf_power=18 pa_gain=1 pwr_idx=1
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut  7:  rf_power=19 pa_gain=1 pwr_idx=2
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut  8:  rf_power=20 pa_gain=1 pwr_idx=3
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut  9:  rf_power=21 pa_gain=1 pwr_idx=4
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut 10:  rf_power=22 pa_gain=1 pwr_idx=5
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut 11:  rf_power=23 pa_gain=1 pwr_idx=6
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut 12:  rf_power=24 pa_gain=1 pwr_idx=7
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut 13:  rf_power=25 pa_gain=1 pwr_idx=9
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut 14:  rf_power=26 pa_gain=1 pwr_idx=11
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 txlut 15:  rf_power=27 pa_gain=1 pwr_idx=14
2021-11-01 12:46:07.655 [RAL:VERB] SX1302 rssi_tcomp: coeff_a=0.000 coeff_b=0.000 coeff_c=20.410 coeff_d=2162.560 coeff_e=0.000
2021-11-01 12:46:07.656 [RAL:VERB] SX1302 rxrfchain 0: enable=1 freq=922.2MHz rssi_offset=-215.399994 type=5 tx_enable=1
2021-11-01 12:46:07.656 [RAL:VERB] SX1302 rxrfchain 1: enable=1 freq=923.2MHz rssi_offset=-215.399994 type=5 tx_enable=0
2021-11-01 12:46:07.656 [RAL:VERB] SX1302 ifchain  0: enable=1 rf_chain=0 freq=-200000 bw=0 SF=0 sync_word=0/0
2021-11-01 12:46:07.656 [RAL:VERB] SX1302 ifchain  1: enable=1 rf_chain=0 freq=0 bw=0 SF=0 sync_word=0/0
2021-11-01 12:46:07.656 [RAL:VERB] SX1302 ifchain  2: enable=1 rf_chain=0 freq=0 bw=0 SF=0 sync_word=0/0
2021-11-01 12:46:07.656 [RAL:VERB] SX1302 ifchain  3: enable=1 rf_chain=0 freq=400000 bw=0 SF=0 sync_word=0/0
2021-11-01 12:46:07.656 [RAL:VERB] SX1302 ifchain  4: enable=1 rf_chain=1 freq=-400000 bw=0 SF=0 sync_word=0/0
2021-11-01 12:46:07.656 [RAL:VERB] SX1302 ifchain  5: enable=1 rf_chain=1 freq=-200000 bw=0 SF=0 sync_word=0/0
2021-11-01 12:46:07.656 [RAL:VERB] SX1302 ifchain  6: enable=1 rf_chain=1 freq=0 bw=0 SF=0 sync_word=0/0
2021-11-01 12:46:07.656 [RAL:VERB] SX1302 ifchain  7: enable=1 rf_chain=1 freq=200000 bw=0 SF=0 sync_word=0/0
2021-11-01 12:46:07.656 [RAL:VERB] SX1302 ifchain  8: enable=1 rf_chain=0 freq=-100000 bw=5 SF=7 sync_word=0/0 [STD] Explicit header
2021-11-01 12:46:07.656 [RAL:VERB] SX1302 ifchain  9: enable=1 rf_chain=0 freq=0 bw=0 SF=0 sync_word=0/0
2021-11-01 12:46:07.656 [RAL:VERB] Station device: /dev/spidev0.0 (PPS capture disabled)
2021-11-01 12:46:07.656 [RAL:ERRO] Concentrator start failed: lgw_start
2021-11-01 12:46:07.656 [RAL:ERRO] ral_config failed with status 0x08
2021-11-01 12:46:07.657 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2021-11-01 12:46:07.657 [AIO:DEBU] [3] ws_close reason=1000
2021-11-01 12:46:07.657 [AIO:DEBU] Echoing close - reason=1000
2021-11-01 12:46:07.895 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2021-11-01 12:46:07.895 [AIO:DEBU] [3] WS connection shutdown...
2021-11-01 12:46:07.896 [TCE:VERB] Connection to MUXS closed in state -1
2021-11-01 12:46:07.896 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)
ERROR: UNKNOWN TX STATUS 0x00
ERROR: UNKNOWN TX STATUS 0x00
ERROR: failed to close I2C device (err=-1)
2021-11-01 12:46:17.900 [any:INFO] ./tc.trust:
```

Please check station.conf
```
{
    /* If slave-X.conf present this acts as default settings */
    "SX1302_conf": {                 /* Actual channel plan is controlled by server */
        "device": "/dev/spidev0.0",
        "lorawan_public": true,      /* is default */
        "clksrc": 0,                 /* radio_0 provides clock to concentrator */
        "full_duplex": false,
        "radio_0": {
            /* freq/enable provided by LNS - only HW specific settings listed here */
            "type": "SX1250",
            "rssi_offset": -215.4,
            "rssi_tcomp": {"coeff_a": 0, "coeff_b": 0, "coeff_c": 20.41, "coeff_d": 2162.56, "coeff_e": 0},
            "tx_enable": true,
            "antenna_gain": 0,           /* antenna gain, in dBi */
            "tx_gain_lut":[
                {"rf_power": 12, "pa_gain": 0, "pwr_idx": 15},
                {"rf_power": 13, "pa_gain": 0, "pwr_idx": 16},
                {"rf_power": 14, "pa_gain": 0, "pwr_idx": 17},
                {"rf_power": 15, "pa_gain": 0, "pwr_idx": 19},
                {"rf_power": 16, "pa_gain": 0, "pwr_idx": 20},
                {"rf_power": 17, "pa_gain": 0, "pwr_idx": 22},
                {"rf_power": 18, "pa_gain": 1, "pwr_idx": 1},
                {"rf_power": 19, "pa_gain": 1, "pwr_idx": 2},
                {"rf_power": 20, "pa_gain": 1, "pwr_idx": 3},
                {"rf_power": 21, "pa_gain": 1, "pwr_idx": 4},
                {"rf_power": 22, "pa_gain": 1, "pwr_idx": 5},
                {"rf_power": 23, "pa_gain": 1, "pwr_idx": 6},
                {"rf_power": 24, "pa_gain": 1, "pwr_idx": 7},
                {"rf_power": 25, "pa_gain": 1, "pwr_idx": 9},
                {"rf_power": 26, "pa_gain": 1, "pwr_idx": 11},
                {"rf_power": 27, "pa_gain": 1, "pwr_idx": 14}
            ]
        },
        "radio_1": {
            "type": "SX1250",
            "rssi_offset": -215.4,
            "rssi_tcomp": {"coeff_a": 0, "coeff_b": 0, "coeff_c": 20.41, "coeff_d": 2162.56, "coeff_e": 0},
            "tx_enable": false
        }
        /* chan_multiSF_X, chan_Lora_std, chan_FSK provided by LNS */
    },
    "station_conf": {
        "radio_init": "rinit.sh",
        "RADIO_INIT_WAIT": "5s",
        "RX_POLL_INTV": "10ms",
        "TC_TIMEOUT": "360s",
        "log_file":  "stderr",   /* "station.log" */
        "log_level": "DEBUG",  /* XDEBUG,DEBUG,VERBOSE,INFO,NOTICE,WARNING,ERROR,CRITICAL */
        "log_size":  10000000,
        "log_rotate":  3
    }
}
```

If you notice anything, please give me some advice again.

closed 2022-01-31 by beitler

GitUser8675309, 2024-05-01:
Same problem, cant figure it out for the life of me.

beitler, 2024-05-01:
> Same problem, cant figure it out for the life of me.

Can you please provide logs?