lorabasics/basicstation #137: Unable to set maximum TX Power

danigian, 2021-11-12:
Having a look at the source code, apparently router_config can include a "max_eirp" field.

I have tried, unsuccessfully, the following:

- Setting region "EU863" and the field "max_eirp". This is not working as expected, as I would expect max_eirp to override the maximum power for downlink transmissions. Apparently (and confirmed by source code) max_eirp is completely ignored here, as the config for EU863 is going to pre-set the "txpow" and "txpow2"
- Removing "region" field from my router_config and setting "max_eirp". Regardless of the "max_eirp" field, I would say that the concentrator is always sending with the same power as, on the leaf device, I cannot see any noticeable difference in RSSI or SNR.

Can you please provide any guidance on how to lower the TX Power of my gateway?

danigian, 2021-11-15:
Hi @beitler and thanks for your answer 👍 

Regarding the max_eirp field, I think that especially for testing purposes it might be interesting to have a value allowing you to "limit" the default capabilities of the board.

Regarding your second question, I was able to observe that, irrespective of the value in txpow, the concentrator seems to always emit with the same power (given the RSSI and SNR values I get when I receive an ACK on my leaf device).
Unfortunately, I don't have any other way, than looking into logs, to check what the emitted power is.

danigian, 2021-11-23:
Hi @beitler, any idea on how to solve my issue?

Another test I did was to manually set the following entry as unique txlut entry: {"rf_power": -128, "pa_gain": 0, "dac_gain": 3, "mix_gain":  8, "dig_gain": 0}. In this way I could notice a decrease in RSSI (approaching -120db) and in SNR (very close to 0)

Am I correct in assuming that the currently unique way to lower TX Power for EU is to set these txlut entries?

beitler, 2021-11-26:
> Am I correct in assuming that the currently unique way to lower TX Power for EU is to set these txlut entries?

This approach is not recommended. `TX_GAIN_LUT` is a calibration table which is set during manufacturing and remains constant during the lifetime of the gateway. Instead, the field `router_config.radio_conf.radio_0.txpow_adjust` ([doc](https://doc.sm.tc/station/gw_v1.5.html?highlight=txpow_adjust#rfconf-object),[code](https://github.com/lorabasics/basicstation/blob/60fd9788763ea11ec80ec1b1f683c0374792f71e/src/sx130xconf.c#L118)) can be used to compensate the TX power for antenna gain and cable loss. The value is subtracted from the default TX power.

danigian, 2021-12-16:
Thanks for the explanation @beitler
We will try to make use of txpow_adjust

closed 2021-12-16 by danigian

sp193, 2025-01-03:
> As you point out correctly, the `max_eirp` field is only used when no known region identifier is given. However, this is a legacy feature. The recommended approach is to use one of the well known region identifiers. What is your use case for deviating from the regional parameters?

Hi.

I bring this up as we have a usecase for adjusting the downlink transmission EIRP. I am from Singapore, which falls under a variant of AS923 (now AS923-1). The whole of AS923 is a blanket "region" that covers territories that have ISM bands that meet its criteria. There will be different local regulations.

With the existing logic, we will be forced to have 13dBm (?) downlink transmission power, but our local regulations permit us to use up to 27dBm. It is impossible to increase the downlink transmission power by specifying max_eirp, because the existing logic only permits us to lower it. Unless we create a completely new region that is not named "AS923-1".

What is the intended behaviour, for us to use a higher downlink transmission power than the default? Do we always create new regions and specify max_eirp? Or should we be expecting max_eirp to override the region's default downlink transmission power?

Kerlink's own version of BasicStation also has similar shortcomings and Chirpstack doesn't seem to support changing the downlink EIRP yet. So, I thought of asking here (the reference design's repository) first about the intended design behaviour before I create further tickets elsewhere.

Thanks in advance.

sp193, 2025-01-16:
Hi, hello. As it concerns how BasicStation could be used properly in deployments, how can I get a solution going? I can create a patch, if necessary.

Personally, I feel that "AS923" should always be the region ID, but you should be allowed to specify the appropriate max_eirp value as an override. Aside from my usecase, I suppose it will also serve the usecase where you aren't from Asia, but you want to use AS923 because it aligns with your local regulations.

If I directly went through Kerlink for this, I am concerned that it'll create fragmentation because Kerlink has their own implementation of this protocol.