lorabasics/basicstation #138: SX1301 configuration parsing is inconsistent

adriansmares, 2021-12-03:
The parsing of FSK channels in the SX1301 configuration objects seems to be inconsistent. 

The `bandwidth` field is considered valid for 1.5 reference designs, both per [`IFCONF` object definition](https://doc.sm.tc/station/gw_v1.5.html#ifconf-object), and per parser (`J_bandwidth` field):

https://github.com/lorabasics/basicstation/blob/60fd9788763ea11ec80ec1b1f683c0374792f71e/src/sx1301v2conf.c#L225-L231

However, for 2.0 reference designs, while the [`CHANCONF` object definition](https://doc.sm.tc/station/gw_v2.html#chanconf-object) allows for the `bandwidth` to be present, the parser does not:

https://github.com/lorabasics/basicstation/blob/60fd9788763ea11ec80ec1b1f683c0374792f71e/src/sx1301v2conf.c#L513-L521

This causes `station` to reject the configuration from the LNS if it contains such a `bandwidth` definition.

beitler, 2021-12-06:
Hi @adriansmares ,

Basics Station allows to configure hardware-specific settings in the static `station.conf` file. The structure of these files is specific to each reference design ([v1.5](https://doc.sm.tc/station/gw_v1.5.html), [v2](https://doc.sm.tc/station/gw_v2.html), and [corecell](https://doc.sm.tc/station/gw_corecell.html)). The `sx1301_conf` object inside the `router_config` message defines the channel plan in a generic way but, for historic reasons, closely resembles v1.5 semantics. The definition of the `sx1301_conf` object is given as part of the [LNS protocol spec](https://doc.sm.tc/station/tcproto.html#router-config-message).

The LNS shall adhere to that `sx1301_conf` object definition. Consequently, the LNS shall not set the `bandwidth` parameter. Why are you interested in setting the `bandwidth` field from the LNS?

closed 2021-12-06 by adriansmares

beitler, 2021-12-06:
Thanks, @adriansmares , for the feedback. Going over the docs again I can certainly understand why you reached that conclusion. The documentation is not as clear as it could be on that point.