lorabasics/basicstation #139: Support for single channel gateway?

pterodragon, 2021-12-06:
Am I correct to assume that this LoRaWAN implementation is LoRaWAN specification compliant and only supports multiple channel gateway and there is no way to configure the station to work as a single channel gateway unless the code is modified to fit?

beitler, 2021-12-06:
Hi @pterodragon ,

Basics Station has the flexibility to be configured with any number of channels that are physically supported by the sx130x baseband circuit. This includes a configuration consisting of just a single channel. 
However, I suppose your question is targeted towards other types of 'gateway' platforms which rely on a sx126x or sx127x device chip as baseband processor. Is this right? In that case, station will not be able to support such a setup unless it is ported to a corresponding HAL implementation. All currently support HALs are sx130x-based.
Having said that, it is important to note that single channel gateways are indeed not LoRaWAN compliant, as you point out correctly.

pterodragon, 2021-12-06:
I am using an sx1302 chip. We just want to make sure that by default the station is using multi channel gateway. And as you mentioned, the `station.conf` can be configured to any number of channels. Could you point me to the portion on where this is stated in the configuration file? Is it by number of entries of `tx_gain_lut`?

beitler, 2021-12-06:
The configuration paradigm in station is a bit different than what people are used to in the packet forwarder. The configuration is split between the local `station.conf` ([doc](https://doc.sm.tc/station/conf.html#configuration-files)) file and the `router_config` ([doc](https://doc.sm.tc/station/tcproto.html#router-config-message)) structure sent down by the LNS.
`station.conf` is used to set static configuration which does not change over the life time of the gateway (which rf_chain to use for TX, RSSI offsets, etc).
`router_config` is used to define dynamic settings, like the allocation of channels. 

So, in your case, you'd have to define your channels in the `router_config` portion according to the documentation.

I'm eager to understand your use case for a single channel setup. Is it just a particular testing scenario, or are you pursuing a production installation with this setup? Having a sx1302 at hand, what keeps you from using all its capabilities?

pterodragon, 2021-12-06:
Thank you for your quick reply. My company does not plan to have a single channel setup, but we are evaluating this LoRaWAN gateway implementation and most importantly if it supports multi-channel configuration in some ways. Looks like as you said we should configure the channels on the LNS.

We might be using The Things Network and I am looking at the relevant configurations in TTN and the station configuration.

closed 2021-12-06 by pterodragon