lorabasics/basicstation #142: Can't build solution because of misconfigured mbedtls dependency

danigian, 2022-01-06:
When trying to build the basic station binaries, a failure in cloning from mbedtls dependency is happening.

`mkdir -p build-local/bin
gcc -std=gnu11 -Isrc -DCFG_prog_crc32 src/crc32.c -o build-local/bin/crc32
mkdir -p build-local/bin
gcc -std=gnu11 -Isrc -DCFG_prog_genkwcrcs src/genkwcrcs.c -o build-local/bin/genkwcrcs
build-local/bin/genkwcrcs $(cat src/kwlist.txt | sed -e '/^#/d;s/[ \t]\+#.*//') > build-local/temp-kwcrc.h
mv build-local/temp-kwcrc.h src/kwcrc.h
platform=linux variant=std make -C deps/mbedtls
make[1]: Entering directory '/basicstation/deps/mbedtls'
platform=linux variant=std ./prep.sh
Cloning into 'git-repo'...
warning: Could not find remote branch mbedtls-2.6 to clone.
fatal: Remote branch mbedtls-2.6 not found in upstream origin
make[1]: *** [makefile:55: platform-linux] Error 128
make[1]: Leaving directory '/basicstation/deps/mbedtls'`

The reason for this is a rename in mbedtls repository that moved branch mbedtls-2.6 to archive/mbedtls-2.6

beitler, 2022-01-06:
Hi @danigian,

thanks a lot for bringing this to our attention. While your PR is indeed an acceptable work around for this issue, unfortunately I will not be able to merge it at this time.
We are working on a solution and will update the repo in due course.

Thanks again!

closed 2022-01-06 by beitler

danigian, 2022-01-06:
@beitler I see you fixed the issue. Do you also have any plan to recreate v2.0.5 tag with the fix?

beitler, 2022-01-06:
The tagged release v2.0.5 will stay as it is.
We are planning to push the next tagged release v2.0.6 with numerous more fixes within a few weeks.

commit 6675c77e (2022-01-06): Update mbedTLS dependency to version 2.7.5. Fixes #142