lorabasics/basicstation #145: CUPS documentation issue

danigian, 2022-01-10:
While doing some tests with the CUPS protocol, I realized that there might be a potential issue in the documentation regarding CUPS.

Specifically, at [this page](https://doc.sm.tc/station/cupsproto.html#http-post-response) it is reported that the field 'sigLen' is the 'Length of signature for update blob', but Basic Station will fail if this value is not increased of 4bytes (given [this row](https://github.com/lorabasics/basicstation/blob/6675c77ee7fa9bab86cc8fc2ca8a58fb106ea322/src/cups.c#L370) in the source code)

If sigLen should be increased of 4bytes, could you please update documentation properly?

beitler, 2022-01-10:
Thanks for reporting this issue. 

Indeed the `sigLen` value represents the length of the signature segment plus the 4 bytes preceding the signature segment which contains the identifier of the public key used for signature verification.

The documentation indeed needs to point out this fact.

beitler, 2022-01-31:
This has been fixed [in the docs](https://doc.sm.tc/station/cupsproto.html#http-post-response). Thanks!

closed 2022-01-31 by beitler