lorabasics/basicstation #153: Cannot connect TTN

oziesin, 2022-02-16:
Hi , i am trying to connect my gateway to TTN. I created necessery files like below maybe it was be incomplete.
[image]

i created .key files and .trust file by ttn guide.

my station.conf file is this way
station.conf
`{
    /* If slave-X.conf present this acts as default settings */
    "SX1301_conf": {		     /* Actual channel plan is controlled by server */
	"lorawan_public": true,      /* is default */
        "clksrc": 1,		     /* radio_1 provides clock to concentrator */
	/* path to the SPI device, un-comment if not specified on the command line e.g., RADIODEV=/dev/spidev0.0 */
	"device": "/dev/spidev0.1",  
	/* freq/enable provided by LNS - only HW specific settings listed here */
	"radio_0": {
	    "type": "SX1257",
	    "rssi_offset": -166.0,
	    "tx_enable": true,
	    "antenna_gain": 0
	}
    },
    "station_conf": {
	"log_file":  "stderr", /* "station.log" */
	"log_level": "DEBUG",  /* XDEBUG,DEBUG,VERBOSE,INFO,NOTICE,WARNING,ERROR,CRITICAL */
	"log_size":  10000000,
	"log_rotate":  3,
	"CUPS_RESYNC_INTV": "1s",
	 /*"device": "/dev/spidev0.1",*/
        "radio_init": "rinit.sh",
        "RADIO_INIT_WAIT": "2s",
        "RX_POLL_INTV": "10ms",
        "TC_TIMEOUT": "360s"
    }
}`
tc.uri file contains wss://eu1.cloud.thethings.network:443

I coppied all files to my device and i run .station and i faced screen below
[image]

How could i fix this and connect my gateway to TTN server ?

beitler, 2022-02-16:
HTTP status code 403 points to an authentication problem.
Did you register your gateway on TTN using the proper gateway EUI (as printed in your logs):
[image]

Please consult with TTN as for the reasons of the authentication issue.

oziesin, 2022-02-16:
Yes, actually  it works with packet forwarder and chirpstack gateway bridge currently, but i wanna transition to basic station.

beitler, 2022-02-17:
Closing issue, since this is most probably not a bug.

closed 2022-02-17 by beitler