lorabasics/basicstation #154: RAK2247 USB - [HAL:ERRO] [lgw_start:742] FAIL TO CONNECT BOARD

chron0, 2022-02-18:
Ahoy all, first of all, thanks for the new stack. I wanted to move from the old udp-forwarder (which seems to work) to basicstation, unfortunately, my experiments so far have failed to get it to work with it:

RAK2247 USB mpcie module -> mpcie USB pcb -> USB -> x86_64 Linux Host

```
[7364802.777002] usb 1-6: new high-speed USB device number 26 using xhci_hcd
[7364802.903384] usb 1-6: New USB device found, idVendor=0403, idProduct=6010, bcdDevice= 7.00
[7364802.903386] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[7364802.903388] usb 1-6: Product: Dual RS232-HS
[7364802.903389] usb 1-6: Manufacturer: FTDI
[7364802.905361] ftdi_sio 1-6:1.0: FTDI USB Serial Device converter detected
[7364802.905386] usb 1-6: Detected FT2232H
[7364802.905496] usb 1-6: FTDI USB Serial Device converter now attached to ttyUSB0
[7364802.905659] ftdi_sio 1-6:1.1: FTDI USB Serial Device converter detected
[7364802.905680] usb 1-6: Detected FT2232H
[7364802.905782] usb 1-6: FTDI USB Serial Device converter now attached to ttyUSB1
```

I couldnt find any guide or specific configuration about the RAK2247 USB, most of the docs or examples seem to focus on the SPI and PI configurations. This is my current station.conf:

```
{
   "SX1301_conf": {                  /* Actual channel plan is controlled by the server */
       "lorawan_public": true,      /* is default */
       "clksrc": 1,                 /* radio_1 provides clock to concentrator */
       "device": "/dev/ttyUSB0",          /* default SPI device is platform specific */
       "radio_0": {
           /* freq/enable provided by LNS - only hardware-specific settings are listed here */
           "type": "SX1257",
           "rssi_offset": -166.0,
           "tx_enable": true,
           "antenna_gain": 0
       },
       "radio_1": {
           "type": "SX1257",
           "rssi_offset": -166.0,
           "tx_enable": false
       }
       /* chan_multiSF_X, chan_Lora_std, chan_FSK provided by LNS */
   },
   "station_conf": {
     "routerid":    "B00BCAFEDEADBEEF",
     "log_file":    "stderr",
     "log_level":   "DEBUG",
     "log_size":    10e6,
     "log_rotate":  3
   }
}
```

which results in failure to communicate with the RAK2247:

```
2022-02-18 11:18:46.910 [TCE:VERB] Connected to MUXS.
2022-02-18 11:18:46.938 [RAL:INFO] Lora gateway library version: Version: 5.0.1;
2022-02-18 11:18:46.946 [RAL:INFO] [LGW lgw1] clksrc=1 lorawan_public=1
2022-02-18 11:18:46.946 [RAL:INFO]  RX/TX RF0:    867.5MHz rssi_offset=-166.0 type=2 tx_notch_freq=0
2022-02-18 11:18:46.946 [RAL:INFO]  RX    RF1:    868.5MHz rssi_offset=-166.0 type=2 tx_notch_freq=0
2022-02-18 11:18:46.946 [RAL:INFO]  [mSF]   0:    868.1MHz rf=1 freq=-400.0 datarate=0
2022-02-18 11:18:46.946 [RAL:INFO]  [mSF]   1:    868.3MHz rf=1 freq=-200.0 datarate=0
2022-02-18 11:18:46.946 [RAL:INFO]  [mSF]   2:    868.5MHz rf=1 freq=  +0.0 datarate=0
2022-02-18 11:18:46.946 [RAL:INFO]  [mSF]   3:    867.1MHz rf=0 freq=-400.0 datarate=0
2022-02-18 11:18:46.946 [RAL:INFO]  [mSF]   4:    867.3MHz rf=0 freq=-200.0 datarate=0
2022-02-18 11:18:46.946 [RAL:INFO]  [mSF]   5:    867.5MHz rf=0 freq=  +0.0 datarate=0
2022-02-18 11:18:46.946 [RAL:INFO]  [mSF]   6:    867.7MHz rf=0 freq=+200.0 datarate=0
2022-02-18 11:18:46.946 [RAL:INFO]  [mSF]   7:    867.9MHz rf=0 freq=+400.0 datarate=0
2022-02-18 11:18:46.946 [RAL:INFO]  [STD]   8:    868.3MHz rf=1 freq=-200.0 datarate=2 bw=2 
2022-02-18 11:18:46.946 [RAL:INFO]  [FSK]   9:    868.8MHz rf=1 freq=+300.0 datarate=50000 bw=0 sync_word=0/0
2022-02-18 11:18:46.946 [RAL:INFO] SX130x LBT not enabled
2022-02-18 11:18:46.946 [RAL:INFO] Station device: /dev/ttyUSB0 (PPS capture disabled)
2022-02-18 11:18:46.946 [HAL:ERRO] [lgw_start:742] FAIL TO CONNECT BOARD
2022-02-18 11:18:46.946 [RAL:ERRO] Concentrator start failed: lgw_start
2022-02-18 11:18:46.946 [RAL:ERRO] ral_config failed with status 0x08
2022-02-18 11:18:46.946 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2022-02-18 11:18:46.946 [AIO:DEBU] [3] ws_close reason=1000
2022-02-18 11:18:46.946 [AIO:DEBU] Echoing close - reason=1000
2022-02-18 11:18:47.014 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2022-02-18 11:18:47.014 [AIO:DEBU] [3] WS connection shutdown...
2022-02-18 11:18:47.014 [TCE:VERB] Connection to MUXS closed in state -1
2022-02-18 11:18:47.014 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)

```

I am not sure if this is a bug, or if the RAK2247 is not supported over USB or if my configuration is wrong, therefore I would appreciate any hints/ideas very much to finally get this to work with basicsation :)

beitler, 2022-02-18:
hi @chron0 ,

the USB interface of the RAK2247 is implemented via an FTDI USB/SPI bridge. This mode of operation is not supported by the HAL. You will need a custom HAL from RAK. During development phase, you should use the tools provided by the HAL for checking the connectivity to the board.

chron0, 2022-02-18:
> You will need a custom HAL from RAK

Do you know if they provide one? I had a look into 
https://github.com/RAKWireless/rak_common_for_gateway/blob/master/lora/rak2247_usb/loragw_spi.ftdi.c
but atm this seems to be a little over my head without really digging into it...

beitler, 2022-10-06:
Hi @chron0 ,

maybe too late of a follow-up but I was wondering if you found a workaround for the RAK hardware? 
Are you aware of [this project](https://github.com/xoseperez/basicstation) which maintains RAK compatible configurations for LoRa Basics Station?

chron0, 2022-12-04:
@beitler - no i havent yet, checking it out now - just want to run the stack without having to run docker on an arm tho :)
but thank you very much for the ping about it!

chron0, 2022-12-05:
hmm, in that repo there is also a note saying:

NOTE: The basicstation project is not compatible with SX1301 USB LoRa concentrators. This means that you won't be able to use it with a RAK2247-USB.