lorabasics/basicstation #155: Type SX1276 not working.

padaszewski, 2022-03-04:
Hello,

I'm trting to run the ```live-s2.sm.tc``` example on my Raspberry Pi 4b with the ```SX1276``` radio type, [exactly this model](https://wiki.dragino.com/index.php?title=Lora/GPS_HAT).
I configured the ```station.conf``` file with this type. This should work according to [documentation]( https://doc.sm.tc/station/gw_v1.5.html?highlight=sx1276#rfconf-object).

Unfortunatelly I become the following error:
```console
[HAL:ERRO] [lgw_rxrf_setconf:496] NOT A VALID RADIO TYPE
[RAL:ERRO] lgw_rxrf_setconf(0) failed
[RAL:ERRO] Concentrator start failed: lgw_rxrf_setconf
[RAL:ERRO] ral_config failed with status 0x08
```

Do You have any ideas how to get this working?

beitler, 2022-03-04:
Hi @padaszewski ,

the LoRa HAT that you are using has only a single sx1276 (device) chip. It is not a multi-channel, multi-SF gateway design which always has an sx130x chip on the board. Basics Station only supports HALs for sx130x-based multi-channel, multi-SF gateway designs. 

Hence, Basics Station is not compatible with your hardware.

padaszewski, 2022-03-04:
Hi @beitler,

Thank You for the quick reply. I'm quite new into Lora, basically exploring it, so sorry for this kinda newbie question.
In addition could You propose me a library for my HAT which could receive the messages from my nodes?
For the Basic Station I found the product RAK2245 Pi Hat Edition I could use. Unfortunatelly it is sold out and I have to start my project with the HAT I currently have. Would the RAK HAT be a good solution for the future?
The use case is to connect with AWS IoT Services.

beitler, 2022-03-04:
For your RAK Gateways, RAK maintains their own fork of basics station. Please have a look at
https://github.com/mpous/basicstation
https://github.com/xoseperez/basicstation (if you are a docker fan)
https://forum.rakwireless.com/ (RAK forum is a helpful community)
With the RAK you will be able to connect to AWS IoT Core for LoRaWAN .

Concerning single channel gateways, please be aware, that these are not formally LoRaWAN compliant. Network Server providers are dropping support for single channel gateways (see [this post in TTN forums](https://www.thethingsnetwork.org/forum/t/single-channel-packet-forwarders-scpf-are-obsolete-and-not-supported/31117)). That post has some links in the bottom which may contain relevant pointers for your Dragino HAT. Otherwise you should reach out to the manufacturer.

closed 2022-03-04 by padaszewski