lorabasics/basicstation #156: Join Request are not conform to the specification

charlesfendt, 2022-03-25:
Hi,
If I look to the specification, the Join Request need to send the JoinEui and the DeviceEui in the "EUI" format
https://lora-developers.semtech.com/build/software/lora-basics/lora-basics-for-gateways/?url=tcproto.html#lorawan-join-requests
In the Glossary
https://lora-developers.semtech.com/build/software/lora-basics/lora-basics-for-gateways/?url=glossary.html
EUI is defined as en HexString with : separator
Problem: the implementation uses dash!
https://github.com/lorabasics/basicstation/blob/ba4f85d80a438a5c2b659e568cd2d0f0de08e5a7/src/uj.c#L863

Is this an issue in the code or in the documentation?

beitler, 2022-03-25:
Hi @charlesfendt ,

thanks for submitting the issue. Indeed this is a glitch in the documentation. LoRa Basics Station uses dashes (`-`) as byte separators in the representation of EUIs. Hence, the description in the glossary is not correct. It should read:

> A 64 bit number representing a unique identifier. Usually represented in the form `HH-HH-HH-HH-HH-HH-HH-HH` where `HH` are pairs of hex digits.

We will fix this.

charlesfendt, 2022-03-25:
Thank you for the clarification!
We can close the issue after the update of the documentation :-)