lorabasics/basicstation #157: Stop failing on lack of temperature sensor on i.e. RAK5146

ab0oo, 2022-03-29:
Can we please get updated libraries that recognize boards without I2C temperature sensors, and allow the basicstation to start without resorting to hacking or patching the library?
I know these patches exist here:
[https://github.com/xoseperez/basicstation/blob/master/builder/v2.0.5.patch](https://github.com/xoseperez/basicstation/blob/master/builder/v2.0.5.patch)
[https://github.com/xoseperez/basicstation/blob/master/builder/v2.0.6.patch](https://github.com/xoseperez/basicstation/blob/master/builder/v2.0.6.patch)
Can we please get these mainlined? I appreciate Xose's work, but I don't want to run in a docker container for various reasons, and I think it's incumbent on the hardware manufacturer to support ALL of their hardware in the base repo.

Cheers,
John Gorkos

beitler, 2022-03-31:
Hi John,

thanks for submitting the issue. It seems that the RAK5146 is not fully compatible with the corecell reference design. The right approach is then to introduce a new 'platform' definition in basics station for RAK5146 which is derived from corecell platform definition but includes the corresponding patches to the HAL which are needed for RAK5146. 

This is, in fact, the general approach for supporting different platforms which don't the 'vanilla' reference designs. More that topic can be found [in the documentation](https://lora-developers.semtech.com/build/software/lora-basics/lora-basics-for-gateways/?url=compile.html).

ab0oo, 2022-03-31:
Thank you for that explanation.  I'm now clear on why XosePerez did what he did.
Regards!

closed 2022-03-31 by ab0oo