lorabasics/basicstation #160: Beacon frame format is only for LoRaWAN 1.0.3

gregbreen, 2022-05-27:
I have been using BS 2.0.6 just fine with a LoRaWAN 1.0.4 Class A device. However, I tried to use it with a 1.0.4 Class B device and could not lock on to the beacon. Reason being that the beacon frame format has changed in 1.0.4. The `s2e_make_beacon()` function only supports 1.0.3 beacon frame formats.

This is not some big or urgent problem for me, but I'm curious if there's any plans to address this?

beitler, 2022-05-31:
Hi @gregbreen,

thanks for reporting the issue. Can you be more specific on where you see the incompatibility? If I see it correctly, the beacon layout structure in the `router_config` allows to construct 1.0.4 Beacons with `Prec=0` which is the correct default value for on-PPS TX scheduling.

gregbreen, 2022-06-01:
Thanks @beitler . Yes I agree with that. I'm not sure if it's deliberate or just a happy accident that the `memset()` call and time offset together will give Param value of 0. I was expecting to see more explicit support for the precision parameter in the **bcning** command. But yes, now that I change the layout, it works with LoRaMAC-node 4.6.0. Thank you.

beitler, 2022-10-06:
I believe this is not by accident but a deliberate choice of the specification to ensure backwards compatibility w.r.t to the fields which were formerly RFU. 

Please allow me to close the issue as a resolution was found. Thanks!

closed 2022-10-06 by beitler