lorabasics/basicstation #161: Using AWS basic Station,  found TX power to be 0.0dBm

kilight254, 2022-06-22:
After my test, the power of TX will be 12dBm, the lowest power of SX1301,
I guess there is a server configuration problem or there is a problem with my configuration file, Am I missing something important?
2022-06-22 01:57:17.532 [S2E:INFO] TX ::0 diid=39028 [ant#0] - on air: **925.7MHz 0.0dBm** ant#0(0) DR10 SF10/BW500 frame=20309C905190DF9AD2E59C5E..DE87E47C
[image]

[aws_station.txt](https://github.com/lorabasics/basicstation/files/8953964/aws_station.txt)
[station.txt](https://github.com/lorabasics/basicstation/files/8953972/station.txt)

beitler, 2022-06-22:
Hi @kilight254 ,

could you please run station with XDEBUG log level (`-l XDEBUG`). This will print the messages from the LNS. We need the content of the `router_config` message as well as the `dnmsg` message of the downlink in order to identify the issue. 

Thanks!

kilight254, 2022-06-22:
Hi, @beitler [beitler](https://github.com/beitler)

[aws_station.txt](https://github.com/lorabasics/basicstation/files/8955232/aws_station.txt)
Can the TX power of SX1301 be modified in the configuration file?
I looked at the SX1302 Station config, which can do just that.

closed 2022-06-24 by kilight254