lorabasics/basicstation #162: BasicStation does not work with AWS for AS923 due to issue in parsing region_id in router_config

Th3G4mbl3r, 2022-06-25:
Hi,

I am using the current version of basicstation - 2.0.6 to connect against AWS IoT Core for LoRaWAN using OTAA. 

The incoming router_config message structure is of type:
{"msgtype":"router_config","NetID":null,"JoinEUI":null,"region":"AS923","hwspec":"sx1301/1","freq_range":[920000000,923400000],"DRs":[[12,125,0],[11,125,0],[10,125,0],[9,125,0],[8,125,0],[7,125,0],[7,250,0],[0,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0],[-1,0,0]],"sx1301_conf":[{"radio_0":{"enable":true,"freq":922200000},"radio_1":{"enable":true,"freq":923200000},"chan_FSK":{"enable":true},"chan_Lora_std":{"enable":true,

However, in S2E.c - handle_router_config function, the logic does not handle the use case when value of region is 'AS923'. It only handles either 'AS923JP' or 'AS923_1'. This causes the region value to become 0 which then means subsequent dnmsgs are ignored by s2e_onMsg function when it checks for the order of messages received and s2ctx->region==0 scenario.

Once i change the handle_router_config function handle AS923 as a valid region string, everything works perfectly. 

Can i submit a pull request with the small change in code i've made to correctly handle the 'AS923' region scenario with respect to connecting with AWS IoT Core for LoRaWAN?

thanks and regards,
Rohit

Th3G4mbl3r, 2022-06-25:
In fact this issue - https://github.com/lorabasics/basicstation/issues/18 Which is supposed to have been fixed in 2.0.6 is exactly why it is occurring cause you are not accepting the common name for AS923 region.

Th3G4mbl3r, 2022-06-25:
Also based on the documentation at https://lora-developers.semtech.com/build/software/lora-basics/lora-basics-for-gateways/?url=tcproto.html - in the default scenario it is supposed to work in region agnostic mode which i would understand as it will just act as a transparent gateway and just send any downlink packets down to the concentrator ral layer for it to do whatever instead of just dropping it with the error that router_config is not complete yet. So the current implementation is also not in alignment with the product documentation.

Th3G4mbl3r, 2022-07-06:
Any updates on how ca we proceed with this? Btw, this issue will occur for other AS923 sub-regions as well as IN865 as well?

notgeschenk, 2022-07-06:
Hi Rohit!
Thanks for your investigations! We are aware of the mentioned issues. Unfortunately the default behavior for unrecognized regions has changed from 2.0.5 to 2.0.6. The default behavior has already been restored in the master branch and will be included in the next release. We're also planning to include the correct parameters for the so far not supported regions AS923-1, AS923-2, AS923-3, AS923-4, IN865 in the next release.

As a hotfix for your issue you could change code line 1804 in s2e.c from:
`if( s2ctx->region == 0 && (msgtype == J_dnmsg || msgtype == J_dnsched || msgtype == J_dnframe) ) {
`to:
`if( s2ctx->txpow == 0 && (msgtype == J_dnmsg || msgtype == J_dnsched || msgtype == J_dnframe) ) {`

(i.e. replace "`region`" with "`txpow`") 
This way the router_config message will be accepted even without a well-known region.

Hope this helps!
best regards,
Frank

Th3G4mbl3r, 2022-07-07:
Hi Frank (@notgeschenk),

Thanks for the response. I will try to get a copy of the master branch and recompile and test it out once i'm back with my gateway setup. And happy to see that all the currently known regions with the correct common names as listed in RP documents will be supported with the next release.

I will close this issue once i've tested it out. Give me until next week as i am on the road and so don't have access to my gateway.

thanks & regards,
Rohit

beitler, 2022-10-06:
> As a hotfix for your issue you could change code line 1804 in s2e.c from:
> `if( s2ctx->region == 0 && (msgtype == J_dnmsg || msgtype == J_dnsched || msgtype == J_dnframe) ) { `to:
> `if( s2ctx->txpow == 0 && (msgtype == J_dnmsg || msgtype == J_dnsched || msgtype == J_dnframe) ) {`

@notgeschenk This is not the right line reference. 

In fact, the work around is to change this line
https://github.com/lorabasics/basicstation/blob/ba4f85d80a438a5c2b659e568cd2d0f0de08e5a7/src/s2e.c#L1794

to 
```
if( s2ctx->txpow == 0 && (msgtype == J_dnmsg || msgtype == J_dnsched || msgtype == J_dnframe) ) {
```

Here is a patch
```patch
diff --git a/src/s2e.c b/src/s2e.c
index 6a9c009..879d3c9 100644
--- a/src/s2e.c
+++ b/src/s2e.c
@@ -1791,7 +1791,7 @@ int s2e_onMsg (s2ctx_t* s2ctx, char* json, ujoff_t jsonlen) {
         LOG(MOD_S2E|ERROR, "Parsing of JSON message failed - ignored");
         return 1;   // return fail? would trigger a reconnect
     }
-    if( s2ctx->region == 0 && (msgtype == J_dnmsg || msgtype == J_dnsched || msgtype == J_dnframe) ) {
+    if( s2ctx->txpow == 0 && (msgtype == J_dnmsg || msgtype == J_dnsched || msgtype == J_dnframe) ) {
         // Might happen if messages are still queued
         LOG(MOD_S2E|WARNING, "Received '%.*s' before 'router_config' - dropped", D.str.len, D.str.beg);
         return 1;
```

sidpat, 2024-01-12:
Hi @notgeschenk / @beitler , could you please provide some information on official support for AS923 and IN865 regions. Its been a while since the last Basic Station release, and just wondering if this project is still in open development.

krispstack, 2024-09-01:
Hi @beitler ,

Support for IN865 Band

Wvirgil123, 2024-11-05:
@beitler 
Hello, this solution, is the txpow value not assigned accurately?
https://github.com/lorabasics/basicstation/blob/ba4f85d80a438a5c2b659e568cd2d0f0de08e5a7/src/s2e.c#L1033

sp193, 2025-01-03:
14dBm is the default value for downlink transmission power if the region is not recognized. AS923 covers many countries, so even the default of 16dBm may not be the best.
If your LNS supports it, you can specify the correct EIRP via the max_eirp option in router_config.