lorabasics/basicstation #163: Compiling on Raspberry Pi 4 bullseye 64 bit fails

Th3G4mbl3r, 2022-07-17:
Based on the [cross compiling instructions](https://lora-developers.semtech.com/build/software/lora-basics/lora-basics-for-gateways/?url=compile.html#cross-compiling-for-a-specific-target), i've amended setup.gmk as follows to support directly compiling basic station on raspberry pi4 running 64-bit Raspbian Bullseye.

The changes made to setup.gmk are as follows:

```

ARCH.rpi64     = aarch64-linux-gnu
ARCH.corecell64  = aarch64-linux-gnu

CFG.corecell64 = linux lgw1 no_leds sx1302
CFG.rpi64     = linux lgw1 no_leds

DEPS.corecell64 = mbedtls lgw1302
DEPS.rpi64     = mbedtls lgw

LIBS.corecell64 = -llgw1302  ${MBEDLIBS}      -lpthread -lrt
LIBS.rpi64     = -llgw   ${MBEDLIBS}      -lpthread

```

However the compilation fails with the following error for the dependency component - `lgw1302` with the following error message:

```
[aarch64-linux-gnu] CC   ral_lgw.o
../../src/ral_lgw.c: In function ‘ral_tx’:
../../src/ral_lgw.c:239:20: error: ‘LGW_LBT_ISSUE’ undeclared (first use in this function)
  239 |         if( err != LGW_LBT_ISSUE ) {
      |                    ^~~~~~~~~~~~~
../../src/ral_lgw.c:239:20: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [../../makefile.s2core:67: ral_lgw.o] Error 1
make[1]: Leaving directory '/home/pi/basicstation/build-corecell64-std/s2core'
make: *** [makefile:39: s-all] Error 2

``` 

This occurs because to compile `lgw1302`, the process of compiling the library requires to run the following shell script located at `<basicstation repo directory>/deps/lgw1302/prep.sh`. This shell script on line 51, does a git patch using the following syntax - `git apply ../${lgwversion}-${platform}.patch`. And this is a problem because for the 64 bit versions that particular patch file does not exist. And so it is not applied and the compile continues until it gets the above error as that #define is missing. It is applied via the patch file - `V2.1.0-corecell.patch` at line 21.

A similar kind of issue occurs even if i try for standard concentrators cards not based on corecell as the compilation process needs to apply patch files which are not existing for rpi64.

Expectation is that based on the documentation link the compilation process should work properly and generate a valid basicstation executable.

Th3G4mbl3r, 2022-07-17:
To complete the compilation process, i had to manually copy the patch files for corecell/rpi variants depending on which one you are building to the new platform based file name. 

the patch files are located in:

```
<basicstation source repository>\deps\lgw\v5.0.1-rpi.patch
<basicstation source repository>\deps\lgw1302\V2.1.0-corecell.patch

```

In my example, i had to do the following:

```
cd <basicstation source repository>\deps\lgw1302\
cp V2.1.0-corecell.patch V2.1.0-corecell64.patch

```

beitler, 2022-10-06:
Good that were able to resolve the issue. Indeed, if you introduce a new `platform` you need to provide the corresponding HAL patch for that new `platform`. 

Please allow me to close the issue, given that a resolution was found. Thanks!

closed 2022-10-06 by beitler