lorabasics/basicstation #167: About the time issue in the log

YuJiahang520, 2022-08-27:
I ported Basics Station on ESP32
But the time in the log is different from what I am running on the raspberry pi
After I sync the system time via NTP, the log shows UTC time normally
But after a while this time will be synced to a very strange value

`2022-08-27 03:04:26.054 [S2E:INFO]   JoinEUI list: 0 entries
2022-08-27 03:04:26.060 [S2E:INFO]   NetID filter: FFFFFFFF-FFFFFFFF-FFFFFFFF-FFFFFFFF
2022-08-27 03:04:26.067 [S2E:INFO]   Dev/test settings: nocca=0 nodc=0 nodwell=0
2022-08-27 03:04:32.006 [S2E:VERB] RX 868.1MHz DR0 SF12/BW125 snr=-11.2 rssi=-136 xtime=0x2D00000066F694 - jreq MHdr=00 JoinEUI=70b3:d57e:d002:6b87 DevEUI=ccf8:8ff:fff5:e1b2 DevNonce=62123 MIC=1722594752
2022-08-27 03:05:08.095 [SYN:INFO] MCU/SX130X drift stats: min: -4.3ppm  q50: -5.7ppm  q80: -6.7ppm  max: -7.6ppm - threshold q90: -7.6ppm
**2022-08-27 03:05:08.098 [SYN:INFO] Mean MCU drift vs SX130X#0: -5.8ppm
2022-08-27 05:46:53.959 [SYN:INFO] Time sync qualities: min=214 q90=242 max=260 (previous q90=2147483647)**
2022-08-27 05:47:00.262 [SYN:VERB] Time sync rejected: quality=244 threshold=242
2022-08-27 05:47:19.174 [SYN:INFO] MCU/SX130X drift stats: min: -4.8ppm  q50: -6.2ppm  q80: -6.2ppm  max: -6.6ppm - threshold q90: -6.2ppm
2022-08-27 05:47:19.176 [SYN:INFO] Mean MCU drift vs SX130X#0: -5.9ppm
2022-08-27 05:47:55.943 [SYN:INFO] Time sync qualities: min=214 q90=242 max=244 (previ`

beitler, 2022-08-29:
Hi @YuJiahang520,

congrats for porting station to ESP32. I'd be interested in your experience during that process. 

The log that you posted, is that from the Raspberry Pi or from the ESP32?

YuJiahang520, 2022-08-30:
> Hi @YuJiahang520,
> 
> congrats for porting station to ESP32. I'd be interested in your experience during that process.
> 
> The log that you posted, is that from the Raspberry Pi or from the ESP32?

The log is for ESP32

Time suddenly changes from here:

2022-08-27 03:05:08.098 [SYN:INFO] Mean MCU drift vs SX130X#0: -5.8ppm
2022-08-27 05:46:53.959 [SYN:INFO] Time sync qualities: min=214 q90=242 max=260 (previous q90=2147483647)

YuJiahang520, 2022-08-30:
I ported both Packet Forwarder and Basics Station
I can sort out the porting process when I have time

beitler, 2022-08-30:
Thanks. Could you share your implementations of `sys_time` and `sys_utc` ?