lorabasics/basicstation #168: Timesync & beacon clarification

Mandur, 2022-09-15:
Hello,

I am trying to make the Basic Station send a beacon signal for class B devices. Based on [this thread](https://github.com/lorabasics/basicstation/issues/98#issuecomment-831204980), I understand that when pps capture is enabled and station_conf is set to "gps" (when using GPS I used the FIFO file solution) or "fuzzy", I should expect to receive [timesync messages](https://lora-developers.semtech.com/build/software/lora-basics/lora-basics-for-gateways/?url=tcproto.html#time-synchronization) originating from the basic station on the network server to synchronize the time between them. However, I don't receive anything in both cases, preventing any beacon frame to be send (error message below). I must be missing something, but I cannot understand what?

[image]

I experienced the above behavior both on basicstation v2.0.5 and v2.0.6. I am using a SX1301 based concentrator (https://www.thethingsnetwork.org/marketplace/product/rhf0m301-lorawan-module)

my station.conf file

``` json
{
    /* If slave-X.conf present this acts as default settings */
    "SX1301_conf": {                 /* Actual channel plan is controlled by server */
        "lorawan_public": true,      /* is default */
        "clksrc": 1,                 /* radio_1 provides clock to concentrator */
        /* path to the SPI device, un-comment if not specified on the command line e.g., RADIODEV=/dev/spidev0.0 */
        /*"device": "/dev/spidev0.0",*/
        /* freq/enable provided by LNS - only HW specific settings listed here */
        "pps":true,
        "radio_0": {
            "type": "SX1257",
            "rssi_offset": -166.0,
            "tx_enable": true,
            "antenna_gain": 0
        },
        "radio_1": {
            "type": "SX1257",
            "rssi_offset": -166.0,
            "tx_enable": false
        }
        /* chan_multiSF_X, chan_Lora_std, chan_FSK provided by LNS */
    },
    "station_conf": {
        "log_file":  "stderr",
        "routerid": "AC1F09FFFE0648AC",
        "log_level": "DEBUG",  /* XDEBUG,DEBUG,VERBOSE,INFO,NOTICE,WARNING,ERROR,CRITICAL */
        "log_size":  10000000,
        "log_rotate":  3,
        "CUPS_RESYNC_INTV": "1s",
        "pps":"fuzzy"
    }
}
```

Thank you for the help.

Mik

beitler, 2022-09-15:
Hi Mik,

RHF0M301 doesn't seem to have a PPS source. What exactly is the setup you are trying to achieve?

Mandur, 2022-09-15:
Thank you for the fast answer!

Hmm, that would then explain it. I am trying to get the basicstation to send a beacon signal with a given gps location in the beacon frame. 

If I understand correctly then for such gateways, the only solution is the timesyncs originating from the Network server that could "push" the time to the basicstations? (As described [here](https://lora-developers.semtech.com/build/software/lora-basics/lora-basics-for-gateways/?url=tcproto.html#time-synchronization in the **Transferring GPS Time** section.

But if i run the same settings on the gateway with a valid pps source it should work, is this correct?

beitler, 2022-09-15:
`Transferring GPS Time` is an advanced feature for niche applications. Generally, you need a PPS signal routed to the SX1301 in order to be able to emit Class B beacons. This means that you need a GPS receiver in your setup.

Your station.conf currently is not properly set up for class B beaconing. Have a look at [this section](https://lora-developers.semtech.com/build/software/lora-basics/lora-basics-for-gateways/?url=conf.html#class-b-beaconing-settings) to understand how to configure basics station for class B beaconing.

Mandur, 2022-09-15:
As written above, I did other trials with station_conf.pps set to "gps" and station_conf.gps : gps.fifo where I output gpgga values. I kept getting these timesyncs issues there. Now I understand that without a valid PPS source, I won't be able to send beaconing signals. I will proceed in trying with another hardware kit, thank you for the help and indications.

closed 2022-09-15 by Mandur

beitler, 2022-09-15:
> As written above, I did other trials with station_conf.pps set to "gps" and station_conf.gps : gps.fifo where I output gpgga values

The nmea output of the GPS receiver does not play a role in the time synchronization mechanism. They are only for informational purpose and station occasionally sends the position coordinates to the LNS. For time synchronization you only need to configure the pps.