lorabasics/basicstation #169: BasicStation unable to connect to Chirpstack

SRagavendar, 2022-09-16:
Hi,
I have ported BasicStation to run on NXP's i.MX platform and initial testing with the simulation using SPI to connect to SX1302 has been successful.
I also have Chirpstack LoRaWAN Server running on another NXP i.MX platform which is in the same local network. Currently I'm facing an issue wherein putting the IP address and port of the Chirpstack Server is not working.
[image]
[Note: There will be extra comments in this log which was to help us to determine the exact reason of the connection refusal.]

We have verified from Chirpstack Server that the websocket is open and ready to accept connections by using  netcat commands to connect directly. 
Is there any way to determine the exact cause of the connection refusal?

beitler, 2022-09-16:
Hi @SRagavendar ,

the best way to start investigating the issue is to do a TCP packet capture using `tcpdump`. `Connection refused (111)` indicates that the host is reachable but the TCP handshake does not succeed. However, this is contrary to your observation that the connection succeeds using netcat. Did you run netcat on the same host on which you are trying to run station?
Could you post two `tcpdump` captures, one when establishing connection via netcat and one when running station. Based on the packet captures we should be see clearly what is going wrong.

SRagavendar, 2022-09-19:
Hi @beitler, thanks for the reply.
I have captured the packets between the same, 
Here is the tcpdump capture with the one while running the station: [IP address-192.168.1.73, Port-3001]
[image]
Here I'm able to see that the system running Chirpstack is sending back Ack packets to BasicStation.

And here is the capture while running netcat: [IP address-192.168.1.73, Port-2889]
[image]
[Note: in this screenshot both station and netcat captures will be there as I tried them both back-to-back]

I have also attached the captures for your reference

SRagavendar, 2022-09-19:
The captures are not uploading properly, so I'm attaching the captures as a drive link:
BasicStation TCPDump capture: https://drive.google.com/file/d/1gT6NneMCMO8REer5jqliwinzI0lUX7be/view?usp=sharing
Netcat TCPDump capture: https://drive.google.com/file/d/1qaW1y2mArUbs13QPXELZRiD7tHMBlHqG/view?usp=sharing

beitler, 2022-09-21:
Thanks for the captures. Indeed the captures indicate that no process is listening on port 3001 on host 192.168.1.73
[image]

I don't quite understand why you let netcat connect to port 2889. My assumption for this experiment would be that you'd try connecting to same port as station is trying to connect (port 3001) just using netcat. If both, station and netcat behave identically, then the destination host would be at fault. To me this really looks like a networking issue and nothing inherently at fault with LoRa Basics Station here.

beitler, 2022-10-06:
Hi @SRagavendar , 

did you have any luck tracing down why the host did not listen on port 3001?

beitler, 2023-05-26:
I hope you were able to track down the issue. I will close the issue due to inactivity. Feel free to re-open if it's still relevant.

closed 2023-05-26 by beitler