lorabasics/basicstation #175: Peroblem with RAK5146

Nileshbhavsar, 2022-12-08:
Hi,

I tried corecell with RAK 5146 but unfortunately failed with AGC fw read/write check failed error. Please check below log.

```
root@rak-gateway:/opt/ttn-station# /opt/ttn-station/basicstation/build-corecell-std/bin/station --radio-init=/opt/ttn-station/reset_gw.sh
2022-12-08 08:59:02.895 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotate=3)
2022-12-08 08:59:02.895 [SYS:INFO] Station Ver : 2.0.6(corecell/std) 2022-12-08 08:58:33
2022-12-08 08:59:02.895 [SYS:INFO] Package Ver : (null)
2022-12-08 08:59:02.895 [SYS:INFO] mbedTLS Ver : 2.28.0
2022-12-08 08:59:02.895 [SYS:INFO] proto EUI   : 0:fc4b:bcbc:4b90       (/sys/class/net/eth0/address)
2022-12-08 08:59:02.895 [SYS:INFO] prefix EUI  : ::1    (builtin)
2022-12-08 08:59:02.895 [SYS:INFO] Station EUI : fc4b:bcff:febc:4b90
2022-12-08 08:59:02.895 [SYS:INFO] Station home: ./     (builtin)
2022-12-08 08:59:02.895 [SYS:INFO] Station temp: /var/tmp/      (builtin)
2022-12-08 08:59:02.896 [SYS:WARN] Station in NO-CUPS mode
2022-12-08 08:59:03.099 [TCE:INFO] Starting TC engine
2022-12-08 08:59:03.110 [AIO:INFO] ./tc.trust:
cert. version     : 3
serial number     : 82:10:CF:B0:D2:40:E3:59:44:63:E0:BB:63:82:8B:00
issuer name       : C=US, O=Internet Security Research Group, CN=ISRG Root X1
subject name      : C=US, O=Internet Security Research Group, CN=ISRG Root X1
issued  on        : 2015-06-04 11:04:38
expires on        : 2035-06-04 11:04:38
signed using      : RSA with SHA-256
RSA key size      : 4096 bits
basic constraints : CA=true
key usage         : Key Cert Sign, CRL Sig2022-12-08 08:59:03.111 [TCE:INFO] Connecting to INFOS: wss://eu1.cloud.thethings.network:8887
2022-12-08 08:59:04.287 [TCE:INFO] Infos: fc4b:bcff:febc:4b90 muxs-::0 wss://eu1.cloud.thethings.network:8887/traffic/eui-FC4BBCFFFEBC4B90
2022-12-08 08:59:04.287 [AIO:DEBU] [3] ws_close reason=1000
2022-12-08 08:59:04.287 [AIO:ERRO] Recv failed: SSL - The peer notified us that the connection is going to be closed
2022-12-08 08:59:04.287 [AIO:DEBU] [3] WS connection shutdown...
2022-12-08 08:59:04.291 [AIO:INFO] ./tc.trust:
cert. version     : 3
serial number     : 82:10:CF:B0:D2:40:E3:59:44:63:E0:BB:63:82:8B:00
issuer name       : C=US, O=Internet Security Research Group, CN=ISRG Root X1
subject name      : C=US, O=Internet Security Research Group, CN=ISRG Root X1
issued  on        : 2015-06-04 11:04:38
expires on        : 2035-06-04 11:04:38
signed using      : RSA with SHA-256
RSA key size      : 4096 bits
basic constraints : CA=true
key usage         : Key Cert Sign, CRL Sig2022-12-08 08:59:04.291 [TCE:VERB] Connecting to MUXS...
2022-12-08 08:59:05.256 [TCE:VERB] Connected to MUXS.
2022-12-08 08:59:05.445 [S2E:WARN] Unrecognized region: IN865 - ignored
2022-12-08 08:59:05.390 [SYS:VERB] /opt/ttn-station/reset_gw.sh: Forked, waiting...
2022-12-08 08:59:05.391 [SYS:DEBU] execvp argv[0]: </opt/ttn-station/reset_gw.sh>
2022-12-08 08:59:05.392 [SYS:DEBU]        argv[1]: </dev/spidev0.0>
2022-12-08 08:59:05.746 [SYS:INFO] Process /opt/ttn-station/reset_gw.sh (pid=2674) completed
2022-12-08 08:59:05.748 [RAL:INFO] Lora gateway library version: Version: 2.1.0;
2022-12-08 08:59:05.807 [RAL:INFO] [LGW sx1302] full_duplex=0 clksrc=1 lorawan_public=1
2022-12-08 08:59:05.808 [RAL:INFO]  RX/TX RF0:    865.2MHz rssi_offset=-166.0 type=2 rssi_tcomp=0.000 0.000 0.000 0.000 0.000
2022-12-08 08:59:05.809 [RAL:INFO]  RX    RF1:  866.385MHz rssi_offset=-166.0 type=2 rssi_tcomp=0.000 0.000 0.000 0.000 0.000
2022-12-08 08:59:05.810 [RAL:INFO]  [mSF]   0: 865.0625MHz rf=0 freq=-137.5 datarate=0
2022-12-08 08:59:05.810 [RAL:INFO]  [mSF]   1: 865.4025MHz rf=0 freq=+202.5 datarate=0
2022-12-08 08:59:05.810 [RAL:INFO]  [mSF]   2:  865.985MHz rf=1 freq=-400.0 datarate=0
2022-12-08 08:59:05.810 [RAL:INFO]  [mSF]   3: 865.2325MHz rf=0 freq= +32.5 datarate=0
2022-12-08 08:59:05.811 [RAL:INFO]  [mSF]   4:  866.185MHz rf=1 freq=-200.0 datarate=0
2022-12-08 08:59:05.811 [RAL:INFO]  [mSF]   5:  866.385MHz rf=1 freq=  +0.0 datarate=0
2022-12-08 08:59:05.811 [RAL:INFO]  [mSF]   6:  866.585MHz rf=1 freq=+200.0 datarate=0
2022-12-08 08:59:05.811 [RAL:INFO]  [mSF]   7:  866.785MHz rf=1 freq=+400.0 datarate=0
2022-12-08 08:59:05.811 [RAL:INFO]  channel 8 disabled
2022-12-08 08:59:05.811 [RAL:INFO]  channel 9 disabled
2022-12-08 08:59:05.811 [RAL:INFO] Station device: spi:/dev/spidev0.0 (PPS capture enabled)
2022-12-08 08:59:05.811 [HAL:INFO] [lgw_com_open:84] Opening SPI communication interface
2022-12-08 08:59:05.812 [HAL:INFO] [lgw_spi_open:88] Setting SPI speed to 2000000
2022-12-08 08:59:05.813 [HAL:INFO] [lgw_connect:1192] chip version is 0x12 (v1.2)
2022-12-08 08:59:06.909 [HAL:ERRO] [sx1302_agc_load_firmware:1187] AGC fw read/write check failed
2022-12-08 08:59:06.910 [HAL:ERRO] [sx1302_radio_calibrate:545] Failed to load calibration fw
2022-12-08 08:59:06.910 [HAL:ERRO] [lgw_start:878] radio calibration failed
2022-12-08 08:59:06.910 [RAL:ERRO] Concentrator start failed: lgw_start
2022-12-08 08:59:06.910 [RAL:ERRO] ral_config failed with status 0x08
2022-12-08 08:59:06.910 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2022-12-08 08:59:06.910 [AIO:DEBU] [3] ws_close reason=1000
2022-12-08 08:59:06.910 [AIO:DEBU] Echoing close - reason=1000
2022-12-08 08:59:07.052 [AIO:DEBU] [3|WS] Server sent close: reason=1000
2022-12-08 08:59:07.052 [AIO:DEBU] [3] WS connection shutdown...
2022-12-08 08:59:07.053 [TCE:VERB] Connection to MUXS closed in state -1
2022-12-08 08:59:07.053 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)
2022-12-08 08:59:17.057 [AIO:INFO] ./tc.trust:
cert. version     : 3
serial number     : 82:10:CF:B0:D2:40:E3:59:44:63:E0:BB:63:82:8B:00
issuer name       : C=US, O=Internet Security Research Group, CN=ISRG Root X1
subject name      : C=US, O=Internet Security Research Group, CN=ISRG Root X1
issued  on        : 2015-06-04 11:04:38
expires on        : 2035-06-04 11:04:38
signed using      : RSA with SHA-256
RSA key size      : 4096 bits
basic constraints : CA=true
key usage         : Key Cert Sign, CRL Sig2022-12-08 08:59:17.057 [TCE:INFO] Connecting to INFOS: wss://eu1.cloud.thethings.network:8887
2022-12-08 08:59:18.226 [TCE:INFO] Infos: fc4b:bcff:febc:4b90 muxs-::0 wss://eu1.cloud.thethings.network:8887/traffic/eui-FC4BBCFFFEBC4B90
2022-12-08 08:59:18.226 [AIO:DEBU] [3] ws_close reason=1000
2022-12-08 08:59:18.226 [AIO:ERRO] Recv failed: SSL - The peer notified us that the connection is going to be closed
2022-12-08 08:59:18.226 [AIO:DEBU] [3] WS connection shutdown...
2022-12-08 08:59:18.230 [AIO:INFO] ./tc.trust:
cert. version     : 3
serial number     : 82:10:CF:B0:D2:40:E3:59:44:63:E0:BB:63:82:8B:00
issuer name       : C=US, O=Internet Security Research Group, CN=ISRG Root X1
subject name      : C=US, O=Internet Security Research Group, CN=ISRG Root X1
issued  on        : 2015-06-04 11:04:38
expires on        : 2035-06-04 11:04:38
signed using      : RSA with SHA-256
RSA key size      : 4096 bits
basic constraints : CA=true
key usage         : Key Cert Sign, CRL Sig2022-12-08 08:59:18.230 [TCE:VERB] Connecting to MUXS...
2022-12-08 08:59:19.219 [TCE:VERB] Connected to MUXS.
2022-12-08 08:59:19.366 [S2E:WARN] Unrecognized region: IN865 - ignored
2022-12-08 08:59:19.369 [SYS:VERB] /opt/ttn-station/reset_gw.sh: Forked, waiting...
2022-12-08 08:59:19.370 [SYS:DEBU] execvp argv[0]: </opt/ttn-station/reset_gw.sh>
2022-12-08 08:59:19.370 [SYS:DEBU]        argv[1]: </dev/spidev0.0>
2022-12-08 08:59:19.713 [SYS:INFO] Process /opt/ttn-station/reset_gw.sh (pid=2691) completed
2022-12-08 08:59:19.713 [RAL:INFO] Lora gateway library version: Version: 2.1.0;
2022-12-08 08:59:19.762 [any:CRIT] Radio device '/dev/spidev0.0' in use by process: 2672

```

Please suggest me for this.

beitler, 2022-12-08:
Hi @Nileshbhavsar, there errors are coming from the HAL (hardware abstractions layer). Basics Station is just a consumer of the HAL APIs. That's why I think you will be best off talking directly with the gateway manufacturer which maintains the HAL in order to understand why you see the errors.

Nileshbhavsar, 2022-12-09:
Hello sir,

Previously I used same setup and it was working fine but right now i doing same process but it is failed to work. Herewith I mentioned steps below which I used.

1). I used this link for RAK 5146: https://github.com/xoseperez/basicstation/tree/master/builder
2). Make same build.sh, v2.0.5.patch and v2.0.6.patch in my gateway.
3). Run build.sh which makes basicstation folder and download all files
4). Change in basicstation/src/s2e.c file because of we are using this in india
5). run "make platform=corecell variant=std"
6). Generate tc.uri, tc.key, tc.trust, and reset_gw.sh file
7). Run the environment using "/opt/ttn-station/basicstation/build-corecell-std/bin/station --radio-init=/opt/ttn-station/reset_gw.sh" 

Please verify once above process if i am doing anything wrong?

techatreyo, 2024-03-05:
> Hi @Nileshbhavsar, these errors are coming from the HAL (hardware abstractions layer). Basics Station is just a consumer of the HAL APIs. That's why I think you will be best off talking directly with the gateway manufacturer which maintains the HAL in order to understand why you see the errors.

Hello,

Can we use this repo in openWRT based chirpstack OS ?

sp193, 2025-01-03:
In 2023, I've successfully used this project with the RAK5146 without issues, using the reference code from this repository. I used the mPCIe card that provides a USB interface. It has a SX1302.

I believe that OP used a different edition, which has the SPI instead. Given that they're all RAK5146, I suppose it should still mean that the device is compatible with the reference BasicStation module.