lorabasics/basicstation #186: gpio reset script issue - stdn mode

StevenResiot, 2023-08-10:
Hi,

we compiled the basicstation `stdn` variant (v1.5 with multiple 1301 boards).
it looks to be working fine

we're now trying to implement the reset script using the `station_conf`.`radio_init` field in `station.conf`

the issue we're having is we have N different scripts ( 1 per board ), so we joined them in a single startup script .
However, it seems the script is being called N times ( 1 per slave-X.conf file ) in parallel, which basically messes up the entire gpio reset procedure.

After noticing this, we tried moving the `station_conf`.`radio_init` field inside the `slave-X.conf` files, at different heights as well, but they're being ignored.

Is there a clean way to do this? One script per slave, or one script only being called only once?
Or should we just give up and call the script before starting basicstation?

Thank you!

beitler, 2023-08-10:
Hi @StevenResiot ,

the intended way to initialize multiple radios with a single initiatlization script in the master/slave configuration of basics station is the following: station calls the configured script, passing the path of the radio device as an argument, e.g.:
```
2023-08-10 13:40:51.148 [SYS:VERB] rinit.sh: Forked, waiting...
2023-08-10 13:40:51.148 [SYS:DEBU] execvp argv[0]: </bin/bash>
2023-08-10 13:40:51.149 [SYS:DEBU]        argv[1]: <rinit.sh>
2023-08-10 13:40:51.149 [SYS:DEBU]        argv[2]: <rinit.sh>
2023-08-10 13:40:51.149 [SYS:DEBU]        argv[3]: </dev/ttyACM0>
```
Hence, you should have a single script which takes the radio device path as input (SPI or serial) and applies the necessary GPIO operations based on the radio device path. 
Is this possible in your case?

StevenResiot, 2023-08-10:
hi @beitler 
it's totally possible, however, how is arg 3 passed/configured?