lorabasics/basicstation #196: SX1302 Integrate with AWS IOT

Meimurugan11, 2024-06-19:
Hi all,
Am using waveshare SX130x LoRaWAN Gateway Module/HAT for rpi 4.
I want to integrate the sx1302 lorawan packet forwarder with aws server it requires aws key, certifates, trust certificates.
I am trying to configure with sample configuration files am not sure this format is right.
Please help me to resolve this issue.Any help would be greatly appreciated.
If possible could you give me the sample station.conf file for aws integration.

Thanks

chaudhariatul, 2024-06-29:
Which OS are you running?

Share output for `arch`, `uname -a` and `cat /etc/os-release`.

KSL-Support, 2024-09-06:
i am in a slightly similar situation, if you could let me know if you have a fix, i am tying to integrate an sx1303(uses the same hal as the 1302 so no difference there) but im using an rpi3b+ instead of a 4, its running debian 11 32bit lite 
as for the outputs
`arch`: armv71
`uname -a`: `Linux lora 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
cat /etc/os-release : 
```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
if you could assist it would be greatly appreciated