lorabasics/basicstation #197: Error in Station EUI if the MAC address contains zero (0)

VictorDenisenko, 2024-06-28:
If Basic Station MAC address contains zero, then Basic Station cannot make EUI from it. 
As for example, if Basic Station has MAC address  
"ether b8:17:eb:f4:05:d4",
then Basic Station make EUI from MAC address like 
"b817:ebff:fef4:5d4"  
instead of 
"b817:ebff:fef4:05d4", 
i.e. the customer cannot use your device if its MAC address contains zero.

beitler, 2024-06-28:
Hi @VictorDenisenko , 

omission of leading zeros is part of the [id6 format spec](https://doc.sm.tc/station/glossary.html#term-ID6). It is inspired by the [IPv6 format](https://en.wikipedia.org/wiki/IPv6#Address_representation) of 128-bit addresses, just scaled down to 64-bit.

VictorDenisenko, 2024-06-28:
I have Dragino PG3102 and trying to connect it to TTN. To get Station EUI, I launch basic station and see their Station EUI containing zero. I cannot change its EUI because it forms from MAC and I cannot change MAC because I am not manufacturer. What I need to do in this case?