lorabasics/basicstation #200: Received 'dnmsg' before 'router_config' - dropped

VictorDenisenko, 2024-08-05:
I have issue "[S2E:WARN] Received 'dnmsg' before 'router_config' - dropped" in Basic Station. Chirp stack LNS receiving Join message from end device and answer it, but Basic station dropped it. End devices cannot connect.
Where and how can I look for the reason? 
Thank you!
Victor

beitler, 2024-08-19:
Hi Victor, 

you are right. IN865 and RU864 are not supported. Unfortunately, specifying unsupported region identifiers leads to a situation where downlinks may be rejected. Please look at [this issue](https://github.com/lorabasics/basicstation/issues/173) and workaround provided there. This should enable you to schedule downlinks.

VictorDenisenko, 2024-08-20:
Hi Anton,
yes, your suggestion works. But it seems that there needs extra changes in Basic Station source code, because it works with TTN, but doesn't with Chirpstack. Chirpstack doesn't  send settings to Basic station. But end device for another region (I tried with KR920) works with Chirpstack and TTN. Below is my log file for RU864 and Basic Station with Chirpstack:

Starting Station ... 
2024-08-19 13:23:18.280 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotate=3)
2024-08-19 13:23:18.280 [SYS:INFO] Station Ver : 2.0.6(corecell/debug) 2024-08-19 06:26:59
2024-08-19 13:23:18.281 [SYS:INFO] Package Ver : 2.0.4
2024-08-19 13:23:18.281 [SYS:INFO] mbedTLS Ver : 2.28.0
2024-08-19 13:23:18.281 [SYS:INFO] proto EUI   : b827:ebff:fef4:16d7	(station.conf)
2024-08-19 13:23:18.281 [SYS:INFO] prefix EUI  : ::1	(builtin)
2024-08-19 13:23:18.281 [SYS:INFO] Station EUI : b827:ebff:fef4:16d7
2024-08-19 13:23:18.281 [SYS:INFO] Station home: ./lns-ttn/	(--home)
2024-08-19 13:23:18.281 [SYS:INFO] Station temp: /var/tmp/	(builtin)
2024-08-19 13:23:18.281 [SYS:WARN] Station in NO-CUPS mode
2024-08-19 13:23:18.482 [TCE:INFO] Starting TC engine
2024-08-19 13:23:18.482 [TCE:INFO] Connecting to INFOS: ws://192.168.0.79:3001
2024-08-19 13:23:18.483 [AIO:XDEB] [3] ws_connecting state=1
2024-08-19 13:23:18.484 [AIO:XDEB] [3] ws_connecting state=2
2024-08-19 13:23:18.484 [AIO:XDEB] [3] socket write bytes=167
2024-08-19 13:23:18.485 [AIO:XDEB] [3] ws_connecting state=3
2024-08-19 13:23:18.485 [AIO:XDEB] [3] socket read  bytes=129
2024-08-19 13:23:18.485 [AIO:XDEB] [3|WS] > {"router":"b827:ebff:fef4:16d7"}
2024-08-19 13:23:18.485 [AIO:XDEB] [3] socket write bytes=38
2024-08-19 13:23:18.486 [AIO:XDEB] [3] socket read  bytes=119
2024-08-19 13:23:18.486 [AIO:XDEB] [3|WS] < {"router":"b827:ebff:fef4:16d7","muxs":"b827:ebff:fef4:16d7","uri":"ws://192.168.0.79:3001/gateway/b827ebfffef416d7"}
2024-08-19 13:23:18.486 [TCE:INFO] Infos: b827:ebff:fef4:16d7 b827:ebff:fef4:16d7 ws://192.168.0.79:3001/gateway/b827ebfffef416d7
2024-08-19 13:23:18.486 [AIO:DEBU] [3] ws_close reason=1000
2024-08-19 13:23:18.486 [AIO:XDEB] [3] ws_closing_w state=5
2024-08-19 13:23:18.486 [AIO:DEBU] Echoing close - reason=1000
2024-08-19 13:23:18.486 [AIO:XDEB] [3] socket write bytes=8
2024-08-19 13:23:18.487 [AIO:DEBU] [3] Connection closed unexpectedly
2024-08-19 13:23:18.487 [AIO:DEBU] [3] WS connection shutdown...
2024-08-19 13:23:18.487 [TCE:VERB] Connecting to MUXS...
2024-08-19 13:23:18.488 [AIO:XDEB] [3] ws_connecting state=1
2024-08-19 13:23:18.488 [AIO:XDEB] [3] ws_connecting state=2
2024-08-19 13:23:18.488 [AIO:XDEB] [3] socket write bytes=180
2024-08-19 13:23:18.491 [AIO:XDEB] [3] ws_connecting state=3
2024-08-19 13:23:18.491 [AIO:XDEB] [3] socket read  bytes=129
2024-08-19 13:23:18.491 [TCE:VERB] Connected to MUXS.
2024-08-19 13:23:18.491 [AIO:XDEB] [3|WS] > {"msgtype":"version","station":"2.0.6(corecell/debug)","firmware":"2.0.4","package":"2.0.4","model":"corecell","protocol":2,"features":"rmtsh"}
2024-08-19 13:23:18.491 [AIO:XDEB] [3] socket write bytes=151
2024-08-19 13:24:18.493 [AIO:XDEB] [3] socket read  bytes=2
2024-08-19 13:24:18.493 [AIO:XDEB] [3|WS] < PING ()
2024-08-19 13:24:18.493 [AIO:XDEB] [3|WS] > PONG
2024-08-19 13:24:18.493 [AIO:XDEB] [3] socket write bytes=6
2024-08-19 13:25:18.495 [AIO:XDEB] [3] socket read  bytes=2
2024-08-19 13:25:18.495 [AIO:XDEB] [3|WS] < PING ()
2024-08-19 13:25:18.495 [AIO:XDEB] [3|WS] > PONG

I also added code in s2e.c
case J_RU864: {
    s2ctx->txpow = 25 * TXPOW_SCALE;
    break;
}
and RU864 to kwlist.txt file.
Can you advise, what else I need to change in Basic Station source code?

beitler, 2024-08-20:
According to your log it looks like the `router_config` message is not received by basics station. According to the LNS handshake procedure the `version` message must be responded to with a `router_config` message:
[image]