lorabasics/basicstation #201: Is the project still active?

feckert, 2024-08-19:
This package is [integrated](https://github.com/openwrt/packages/tree/master/net/basicstation) in the OpenWrt packages feed.
For some time now, mbedtls version 3.x has been used in the master branch of openwrt (https://github.com/openwrt/openwrt/commit/0e06642643862d7b6d1edfd6e0e26dad3eae0cbc).
Without this [pull request](https://github.com/lorabasics/basicstation/pull/198) to fix mbedtls compilation, it is no longer possible to build the `basicstation` packages in OpenWrt.
Since the pull request has not been merged, I have created a [pull request](https://github.com/openwrt/packages/pull/24805) against the OpenWrt packages feed with this change as patch. So that the `basicstation` builts again.
Now the question has [arisen](
https://github.com/openwrt/packages/pull/24805#issuecomment-2294003183) whether it is better to delete the package completely, as nothing is happening here **since two years**!

I would like to ask you to take a position.

beitler, 2024-08-19:
Hi @feckert ,

thank you for the question. The project is still active. It is used in many LoRaWAN gateway products and is supported by almost all LoRaWAN network servers. The lack of code updates since Jan 18, 2022 is not ideal but is mostly a consequence of the fact that there are no open critical bugs or security vulnerabilities on the project. There are many things that could be improved, for sure, but stability was favored over agility for this project which may also have fostered this widespread adoption, don't you think?

Having said that, I think upgrading mbedtls to version 3.x is a valid request.

feckert, 2024-08-21:
@beitler Thank you for your feedback. The pull request to fix the build against mbedtls version 3.x.x has been [added](https://github.com/openwrt/packages/pull/24805) to the master branch. I hope that a new version with the mbedtls 3.x.x fix will be released soon.  So we could remove the patch.

closed 2024-08-21 by feckert